PX4/PX4-Autopilot #16486: error while running make px4_sitl_default gazebo

Iko-Git, 2021-01-04:
We use GitHub issues only to discuss PX4 bugs and new features. For
questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).

Thanks!

Jaeyoung-Lim, 2021-01-17:
Did you go through the toolchain installation as documented in http://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html?

You can execute the script
```
Tools/setup/ubuntu.sh
```

FlavioTonelli, 2021-01-18:
Looks similar to an issue I had. Running ubuntu.sh is probably not enough to fix the issue.
https://github.com/PX4/PX4-Autopilot/issues/16242

Jaeyoung-Lim, 2021-01-18:
@FlavioTonelli Can you confirm that you ran the script on a fresh system? Otherwise if you had a different version of protobuf already installed, that might explain why the script could not install the proper version

FlavioTonelli, 2021-01-18:
No it was not a fresh system. That's the reason why I had to manually fix the version linking on my ubuntu.
That fixed the issue on my side. I suppose @Iko-Git its facing a similar issue

Jaeyoung-Lim, 2021-03-20:
closing as stale. @Iko-Git Please reopen if you think this is still an issue

closed 2021-03-20 by Jaeyoung-Lim