PX4/PX4-Autopilot #16500: VTOL RTL mode

MartinStokelj, 2021-01-06:
**Describe the bug**
If something goes wrong during flying in FW mode or by pressing RTL the drone goes back to WP (launch/home) but instead of loitering around WP he makes transition to MC and starts hover above WP (he does not land, which is ok).
RTL_LAND_DELAY is set to -1.
Param description:
Delay before landing (after initial descent) in Return mode. If set to -1 the system will not land but loiter at RTL_DESCEND_ALT.

**To Reproduce**
Steps to reproduce the behavior:
1. During flying VTOL in FW mode, press RTL.

**Expected behavior**
The drone should go to RTL and loiter around WP as FW, he should not make the transition to MC.
Why? As FW the battery usage is lower so we always prefer to stay it as FW. Since MC and MC landing takes a lot of battery we don't want drone to land if not commanded by drone operator. The drone operator need to decide if he wants to land or continue mission or change mission. If loss of communication is problem I suggest:
- if drone does not have connection, go to land
- if drone has connection stay as FW and loiter around home

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
[ulg link](https://review.px4.io/plot_app?log=1e8988c9-6556-4a91-b766-e14f2674efa0)

**Drone (please complete the following information):**
- SITL vtol standard

moreba1, 2021-01-06:
>  The drone should go to RTL and loiter around WP as FW, he should not make the transition to MC.

you should disable NAV_FORCE_VT for this.

MartinStokelj, 2021-01-07:
> 
> 
> > The drone should go to RTL and loiter around WP as FW, he should not make the transition to MC.
> 
> you should disable NAV_FORCE_VT for this.

It does the same. It goes to MC mode and hover above WP.

sfuhrer, 2021-01-07:
I can check if it's an easy thing to make it work with `NAV_FORCE_VT`.

sfuhrer, 2021-01-18:
@MartinStokelj  have a loot at https://github.com/PX4/PX4-Autopilot/pull/16582

MartinStokelj, 2021-07-16:
@sfuhrer
Today I tested on master with gazebo, I put NAV_FORCE_VT=0, RTL_LAND_DELAY = -1
During critical battery warning drone  went RTL and start loiter arround home location (great, so pilot can decide what he will do (continue mission or land)), 
when he triggered emergency battery warning he said: Transition denied: ARMED to SHUTDOWN and start to descend as FW and crashed on the ground (he should be able to make transition and land normaly as MC).

sfuhrer, 2021-07-20:
Sorry to hear about the crash. Can you share a log?

MartinStokelj, 2021-07-25:
@sfuhrer Sorry for late response, I am on vacation... There was no crash, I tested with gazebo simulator. I thought it will work diferently.
My thinking:
- if drone is in FW mode he should always fly in FW since MC drains battery very fast. So when entering RTL he should loiter arround home location as FW and that was ok.
- if drone triggers land he should land as MC because of limited space or because belly land will damage the drone

sfuhrer, 2021-08-13:
And now it was my turn to be on vacation :)

Ah okay, I missed that it was in gazebo. 

I really wonder what your requirements are. NAV_FORCE_VT=0 means that it will never itself switch to hover before landing. Leaving it at 1 means that in below the Critical level it will RTL, then loiter in FW around home, then if battery is below Emergency it will transition to hover and Land.

closed 2021-08-13 by MartinStokelj