PX4/PX4-Autopilot #16501: Flag experimental boards in the build system

LorenzMeier, 2021-01-06:
We need to make sure experimental boards are more clearly flagged all along the way, according to this list:

https://px4.io/autopilots/

mrpollo, 2021-02-03:
Command-line warnings when building for an experimental board

mrpollo, 2021-02-10:
**TODO**
* Let's give manufacturers fair warning this is happening, and what it takes to fix it before we implement.
* Draft a clear and public timeline
* Let's make sure documentation, website, and source all agree on the same definition of board support status
* Let's add metadata that can help warn users when building for an experimental board
* Let's also add instructions on how to move from category to category to help manufacturers increase PX4 support

cc @jinger26 let's work together on this one

mrpollo, 2021-02-10:
From @MaEtUgR what do we do with older boards that are supported but are on their way out like FMUv2 hardware?

mrpollo, 2021-02-10:
From @davids5, maybe we should introduce LTS hardware, or specify the support window per hardware