PX4/PX4-Autopilot #16510: USB HS support on CUAV X7/Nora

coderkalyan, 2021-01-07:
I don't think this was brought up before, so I'm just filing an issue for us to keep track of it - CUAV X7 and Nora support USB HS, but I don't think this is supported in PX4. It would be nice to have support for this.

I am able to test support for this if anyone is interested. Additionally I can try to look into implementing it but I'm not yet sure what is involved (PX4 usb subsystem? Nuttx? Build setup?)

dagar, 2021-01-07:
We'll need to dig into the NuttX side to see if USB highspeed on stm32h7 is fully supported.

For the new CUAV boards one annoyance we'll have to deal with in PX4 is the conflict with USB HS and CAN2. I think it's workable, but will inevitably burn someone that connects USB to configure/monitor/etc something on CAN through QGC. 

[image]

FYI @davids5

coderkalyan, 2021-01-07:
Ah, that does seem like an issue. @dagar are the CAN1 and CAN2 controlled together in software or something? Or isn't it possible to just disable CAN2 while USB is connected and continue using the CAN1 bus?

dagar, 2021-01-07:
> Ah, that does seem like an issue. @dagar are the CAN1 and CAN2 controlled together in software or something? Or isn't it possible to just disable CAN2 while USB is connected and continue using the CAN1 bus?

We can clobber the configuration in place breaking CAN2 and it will "work". The problem is how that kind of thing will ripple through the system if people are actually using CAN and we try to lock down the system. Is the system complaining because it thinks it lost a CAN bus? Did it have certain peripherals on CAN2 that have now disappeared? What if you have an external magnetometer only on CAN2, plug in USB to calibrate, then once you go to actually use the system it's complaining about an uncalibrated mag.

These likely aren't major issues, but it's an annoyance that I would have preferred to not have to deal with. We can get away with things like this when it's poorly defined (sensors and peripherals that happily come and go), but more and more as we try tightly integrate everything to create a robust system it's going to become problematic. End of rant.

coderkalyan, 2021-01-07:
@dagar ahh ok, makes sense. I forgot that some people may connect separate peripherals to the CAN1 and CAN 2 bus, I was only thinking about the use case where the same peripherals are connected to CAN1 and CAN2 for redundancy. I guess it's something we'll have to deal with though.

dookei, 2021-11-29:
Hi @dagar  and @davids5 any news on the support for USB HS? According to Nuttx menuconfig is already supported.
Anyone gave it a try already ? 
[image]

davids5, 2021-11-29:
@dookei - that is at the high level driver enable. The SoC does not have internal support for HS.  So the HW has to have an external ULPI transceiver. If it does, then there are stm32h7 driver setting and support that need to be enabled and tested. I am not sure of the current driver status.