PX4/PX4-Autopilot #16516: Hold mode without global position

mortenfyhn, 2021-01-08:
It would be useful to allow hold mode flight without a global position estimate (but with a valid local estimate). One use case is indoor flight with the RC loss failsafe delay being implemented in #16406.

I reckon any vehicle that can fly in position mode should be able to fly in hold mode, but the firmware doesn't currently allow it.

I tried to mess around with the logic and allow this, and was eventually able to enter hold mode in simulation with only a local estimate, but the vehicle did not properly hold its position, so I probably didn't dig deep enough.

MaEtUgR, 2021-01-18:
I agree. It's currently not allowed because the logic of hold mode in navigator uses global coordinates partly also because Hold mode is used for gotos which should have its own mode see https://github.com/PX4/PX4-Autopilot/issues/14576

> I probably didn't dig deep enough

Atm you'd need to dig quite deep but the navigation stack changes have the exact goal to make this easily achievable.

See
https://github.com/PX4/PX4-Autopilot/projects/34
https://docs.google.com/document/d/1N59JHFqL_E9X4b-U0CzxW_eydqXUHZMxAH_FTD30ilk/

We're happy if you want to help. I think it would be useful to have a separate hold mode flight task that does the right thing incorporated in a way that doesn't break goto. At least that would be my approach to go about it.

mortenfyhn, 2021-01-20:
I understand, thanks. I can't guarantee I'll dig much deeper soon, though, but let's leave the issue open.