PX4/PX4-Autopilot #16517: Typhoon H480 SITL Gazebo Oscillation Problem

canberkdurmus, 2021-01-08:
**Describe the bug**
Hi, I am having stability issues with Gazebo SITL Typhoon h480. In the hover it gets worse with time, descending and ascending with no major problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the repository with `make px4_sitl gazebo_typhoon_h480`
2. Open QGroundControl and Takeoff to any height
3. Oscillation can be observed, while waiting it gets amplified 

**Expected behavior**
Expected behavior is of course a hover without major oscillations but small tuning errors are always acceptable for simulating the real world conditions. 

**Screenshots**

https://user-images.githubusercontent.com/21319327/104027976-32fb1180-51d9-11eb-8344-c3e1b38ab3ca.mp4

**Drone:**
- Typhoon H480 SITL 

**Proposed Solution**

I have made a rough tuning with [PX4 Multicopter PID Tuning Guide](https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter.html) below you can see the values and flight video. I am going to open a pull request as soon as possible.

https://user-images.githubusercontent.com/21319327/104028300-9dac4d00-51d9-11eb-8ad0-7cb32d296f31.mp4

The tuning is not very good I am aware of it but it can fly fine now.
Pitch:
[image]

Roll:
[image]

Yaw (Not changed I think):
[image]

Thank you

Jaeyoung-Lim, 2021-01-08:
This has been reported also in https://github.com/PX4/PX4-Autopilot/issues/14953

canberkdurmus, 2021-01-08:
I am sorry I did not see this issue. Here is the PR: #16519

canberkdurmus, 2021-01-16:
We can close this issue since it is been resolved in #16519

closed 2021-01-16 by Jaeyoung-Lim