PX4/PX4-Autopilot #16526: Lidar based altitude not working using terrain following

shobhitme2, 2021-01-09:
Hello I have made a drone using VIO for position. I am trying to use a lidar as the primary height source and trying to follow terrain. However the altitude hold drifts up and down, it is more pronounced using terrain following rather than altitude following. In mission planner and APM lidar based altitude hold works great. What might be the issue? Is it a delay parameter?

dagar, 2021-01-09:
Can you share a link to the logs with current master? https://logs.px4.io/

shobhitme2, 2021-01-09:
I was running the stable FW, here is a link. Do you want me to upload the master branch and test?  
https://review.px4.io/plot_app?log=207ef16d-9b23-47b4-98a6-5d705d4c2353

dagar, 2021-01-09:
> Do you want me to upload the master branch and test?

Yes, it's extremely helpful if you can get an EKF2 replay log from boot with current master. https://dev.px4.io/master/en/debug/system_wide_replay.html#ekf2-replay

bresch, 2021-01-27:
relates to https://github.com/PX4/PX4-ECL/issues/953

shobhitme2, 2021-01-27:
I use vision as the primary height source the local position Z seems to be much better and does not drift. My setting are using vision as the EKF2_HGT and setting range sensor ENABLED and ALT_MODE set to terrain following. I going to conduct to conduct a long range test (~200m) over flat terrain with this setting. If the estimator is using vision as the z source the drift will be noticeable, but if the lidar is indeed being utilized it will easy to discern via flight logs and visually.  

[image]
[image]

priseborough, 2023-02-07:
We should not be using the EKF for terrain following in a way that violates the requirement that it's local earth frame is non accelerating and that includes the vertical direction. That means that the 0 height datum should not be moving up and down following terrain.

priseborough, 2023-02-07:
Using Lidar as the EKF's primary height sensor during flight over varying terrain is not recommended.

priseborough, 2023-02-07:
I would recommend using baro or GPS as the primary height source (it will drift slowly) and rely on the position controller to regulate height above ground as reported in the vehicle_local_position.distance_bottom uORB message. Check that the EKF is using LiDAR to update the terrain height offset. If it is then estimator_innovations.HAGL should be varying.

closed 2023-02-07 by priseborough