PX4/PX4-Autopilot #16531: Speed accuracy for blended GPS solution should be weighted average or max of reported

CarlOlsson, 2021-01-11:
I have a question regarding the speed accuracy that we push into ECL in the blended GPS solution.

E.g. a vehicle has two GPS receivers with different speed accuracy. One with significantly less velocity measurement noise and smaller reported speed velocity variance than the other.

Now we set the `s_variance_m_s` to the [minimum of the two receivers](https://github.com/PX4/PX4-Autopilot/blob/967d35a6b69cda58be5154598433ba8ce9afb73c/src/modules/sensors/vehicle_gps_position/VehicleGPSPosition.cpp#L456-L459), but the velocity measurement themselves are the [weighted average of the two](https://github.com/PX4/PX4-Autopilot/blob/967d35a6b69cda58be5154598433ba8ce9afb73c/src/modules/sensors/vehicle_gps_position/VehicleGPSPosition.cpp#L436).

In my opinion this doesn't make full sense. We should either set the accuracy to the weighted average of the reported speed accuracy, or to the maximum of the two.

bresch, 2021-01-26:
That's right, we should improve that. Assuming the measurement are uncorrelated, we could compute the variance of the blended solution using the following formula:
`Var(v) = w_1^2 * Var(v_1) + w_2^2 * Var(v_2)`
Where `w_x`  is the weight of each receiver in the blended solution and `Var(v_x)` is the velocity variance reported by each receiver.

priseborough, 2023-03-07:
The variance weighted average of two velocity measurements returned by the blending algorithm should have an error that is smaller than the either of the two individual measurements on their own, hence the design decision to take the smaller of the two individual speed accuracy values as a conservative estimate.

priseborough, 2023-03-07:
The formula provided by @bresch is the correct calculation for gaussian error sequences with specified variances, but is less conservative.

closed 2023-03-07 by bresch