PX4/PX4-Autopilot #16535: PX4 Parameter for Update Rate of leddarone rangefinder

ashwinvk94, 2021-01-11:
The interval rate at which the data from the [leddarone sensor](https://docs.px4.io/master/en/sensor/leddar_one.html) is[ set to a constant of 100ms(10Hz)](https://github.com/PX4/PX4-Autopilot/blob/master/src/drivers/distance_sensor/leddar_one/LeddarOne.hpp#L52). But the sensor is capable of sending data at much faster rates. This can be adjusted as described in this [comment](https://github.com/leddartech/LeddarSDK/issues/20#issuecomment-705678411). I have found that an **Accumulation exponent of 8** and **oversampling exponent of 3** gives decent performance. This results in an update rate of 35Hz.

The parameter [LEDDAR_ONE_MEASURE_INTERVAL](https://github.com/PX4/PX4-Autopilot/blob/master/src/drivers/distance_sensor/leddar_one/LeddarOne.hpp#L52) could changed to be used as a px4 parameter making it easily editable through qgrouncontrol.

dagar, 2021-01-11:
I would just make the change and not bother with a parameter.

dagar, 2021-01-11:
In the PX4 driver make sure it configures the device for what you expect, and possibly handle any resets (loss of configuration).

ashwinvk94, 2021-01-11:
Got it thanks

closed 2021-01-11 by ashwinvk94