PX4/PX4-Autopilot #16538: Error launching with multi-vehicle setup using Gazebo and ROS

bonellia, 2021-01-12:
**Describe the bug**
I am following the steps below from the [this webpage.](https://dev.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html#build-and-test)

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the latest version of the PX4-Autopilot repository
2. Enter the following commands within the PX4-Autopilot directory:
```bash
git submodule update --init --recursive
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/sitl_gazebo
roslaunch px4 multi_uav_mavros_sitl.launch
```

**Expected behavior**
Gazebo should run with multiple vehicles visible.

**Log Files and Screenshots**
```
[roslaunch][INFO] 2021-01-12 14:04:15,905: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2021-01-12 14:04:15,908: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2021-01-12 14:04:15,908: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'px4', 'multi_uav_mavros_sitl.launch']
[roslaunch][INFO] 2021-01-12 14:04:15,908: roslaunch env is {'LC_NUMERIC': 'tr_TR.UTF-8', 'ROS_DISTRO': 'melodic', 'TERMINATOR_DBUS_PATH': '/net/tenshu/Terminator2', 'QT4_IM_MODULE': 'xim', 'GJS_DEBUG_OUTPUT': 'stderr', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'PKG_CONFIG_PATH': '/opt/ros/melodic/lib/pkgconfig', 'WINDOWPATH': '1', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'LC_PAPER': 'tr_TR.UTF-8', 'ROSLISP_PACKAGE_DIRECTORIES': '', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'kril', 'USER': 'kril', 'PATH': '/opt/ros/melodic/bin:/opt/gcc-arm-none-eabi-9-2020-q2-update/bin:/home/kril/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'XDG_VTNR': '1', 'HOME': '/home/kril', 'CMAKE_PREFIX_PATH': '/opt/ros/melodic', 'DISPLAY': ':0', 'SSH_AGENT_PID': '1353', 'LANG': 'en_US.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'ROS_LOG_FILENAME': '/home/kril/.ros/log/e8f7ebcc-54c5-11eb-b8ab-2cf05d25adcc/roslaunch-kril-ubuntu-24527.log', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'SESSION_MANAGER': 'local/kril-ubuntu:@/tmp/.ICE-unix/1212,unix/kril-ubuntu:/tmp/.ICE-unix/1212', 'LC_MEASUREMENT': 'tr_TR.UTF-8', 'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'TERMINATOR_UUID': 'urn:uuid:b9b31977-e1bb-4c54-97bd-5dede7012c5b', 'QT_ACCESSIBILITY': '1', 'LD_LIBRARY_PATH': '/opt/ros/melodic/lib:/home/kril/PX4-Autopilot/build/px4_sitl_default/build_gazebo', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'CLUTTER_IM_MODULE': 'xim', 'TEXTDOMAIN': 'im-config', 'XMODIFIERS': '@im=ibus', 'GIO_LAUNCHED_DESKTOP_FILE_PID': '20202', 'XDG_RUNTIME_DIR': '/run/user/1000', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'ROS_PYTHON_VERSION': '2', 'USERNAME': 'kril', 'XDG_SESSION_DESKTOP': 'ubuntu', 'SHLVL': '1', 'GIO_LAUNCHED_DESKTOP_FILE': '/usr/share/applications/terminator.desktop', 'ROS_VERSION': '1', 'GTK_IM_MODULE': 'ibus', 'LC_ADDRESS': 'tr_TR.UTF-8', 'PYTHONPATH': '/opt/ros/melodic/lib/python2.7/dist-packages', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'VTE_VERSION': '5202', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'GDMSESSION': 'ubuntu', 'IM_CONFIG_PHASE': '2', 'TEXTDOMAINDIR': '/usr/share/locale/', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'TERMINATOR_DBUS_NAME': 'net.tenshu.Terminator20x1a6021154d881c', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'ROS_PACKAGE_PATH': '/opt/ros/melodic/share:/home/kril/PX4-Autopilot:/home/kril/PX4-Autopilot/Tools/sitl_gazebo', 'GAZEBO_MODEL_PATH': ':/home/kril/PX4-Autopilot/Tools/sitl_gazebo/models', 'XDG_SESSION_ID': '1', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', '_': '/opt/ros/melodic/bin/roslaunch', 'GTK_MODULES': 'gail:atk-bridge', 'LC_IDENTIFICATION': 'tr_TR.UTF-8', 'DESKTOP_SESSION': 'ubuntu', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'XDG_SESSION_TYPE': 'x11', 'OLDPWD': '/home/kril', 'LC_TELEPHONE': 'tr_TR.UTF-8', 'GAZEBO_PLUGIN_PATH': ':/home/kril/PX4-Autopilot/build/px4_sitl_default/build_gazebo', 'LC_MONETARY': 'tr_TR.UTF-8', 'PWD': '/home/kril/PX4-Autopilot', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'ROS_MASTER_URI': 'http://localhost:11311', 'COLORTERM': 'truecolor', 'LC_NAME': 'tr_TR.UTF-8', 'XDG_MENU_PREFIX': 'gnome-', 'LC_TIME': 'tr_TR.UTF-8', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'GJS_DEBUG_TOPICS': 'JS ERROR;JS LOG', 'XDG_SEAT': 'seat0'}
[roslaunch][INFO] 2021-01-12 14:04:15,908: starting in server mode
[roslaunch.parent][INFO] 2021-01-12 14:04:15,908: starting roslaunch parent run
[roslaunch][INFO] 2021-01-12 14:04:15,908: loading roscore config file /opt/ros/melodic/etc/ros/roscore.xml
[roslaunch][INFO] 2021-01-12 14:04:16,205: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2021-01-12 14:04:16,205: loading config file /home/kril/PX4-Autopilot/launch/multi_uav_mavros_sitl.launch
[roslaunch][INFO] 2021-01-12 14:04:16,228: Added node of type [gazebo_ros/gzserver] in namespace [/]
[roslaunch][INFO] 2021-01-12 14:04:16,229: Added node of type [gazebo_ros/gzclient] in namespace [/]
[roslaunch][ERROR] 2021-01-12 14:04:16,230: RLException: while processing /home/kril/PX4-Autopilot/launch/single_vehicle_spawn.launch:
Invalid <arg> tag: Invalid left parenthesis '(' in substitution args [$(find mavlink_sitl_gazebo)/scripts/jinja_gen.py --stdout --mavlink_id=$(eval 1 + arg('ID')) --mavlink_udp_port=$(arg mavlink_udp_port) --mavlink_tcp_port=$(arg mavlink_tcp_port) --gst_udp_port=$(arg gst_udp_port) --video_uri=$(arg video_uri) --mavlink_cam_udp_port=$(arg mavlink_cam_udp_port) $(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf.jinja $(find mavlink_sitl_gazebo)]. 

Arg xml is <arg default="$(find mavlink_sitl_gazebo)/scripts/jinja_gen.py --stdout --mavlink_id=$(eval 1 + arg('ID')) --mavlink_udp_port=$(arg mavlink_udp_port) --mavlink_tcp_port=$(arg mavlink_tcp_port) --gst_udp_port=$(arg gst_udp_port) --video_uri=$(arg video_uri) --mavlink_cam_udp_port=$(arg mavlink_cam_udp_port) $(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf.jinja $(find mavlink_sitl_gazebo)" name="cmd"/>
[roslaunch][ERROR] 2021-01-12 14:04:16,230: The traceback for the exception was written to the log file
[roslaunch][ERROR] 2021-01-12 14:04:16,230: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 332, in main
    p.start()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 289, in start
    self._start_infrastructure()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 238, in _start_infrastructure
    self._load_config()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 144, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
[... cut ...] odels/$(arg vehicle)/$(arg vehicle).sdf.jinja $(find mavlink_sitl_gazebo)" name="cmd"/>

[rospy.core][INFO] 2021-01-12 14:04:16,230: signal_shutdown [atexit]

```

[image]

**Additional context**
I had older version of the repository which is roughly 2 months old. Before I had this issue, I have deleted the older version and cloned the most recent version.  
I have inspected the file history for the launch file (single_vehicle_spawn.launch)  and it seems the changes are relevant to this problem.

bonellia, 2021-01-12:
https://github.com/PX4/PX4-Autopilot/pull/16497

PR above might be relevant @mzahana

sahin52, 2021-01-12:
This problem occured to me too. The older version which I have ( https://github.com/PX4/PX4-Autopilot/commit/e51000c5be901fc53bb2f5f63b82f9b9074316ea ) works fine, 3 drones are seen, but I clone'd the new version
( https://github.com/PX4/PX4-Autopilot/commit/48f125f150547a061ff97a90c6f39b7b6aba8512 ) And the same problem occurs. It doesn't work.

mzahana, 2021-01-12:
@Jaeyoung-Lim was able to test PR https://github.com/PX4/PX4-Autopilot/pull/16497 successfully. Maybe he can comment on that.

Jaeyoung-Lim, 2021-01-12:
@mzahana I tested it on ROS noetic, so probably this is coming from python version differences. Would you be able to adjust it to also work with python2?

Jaeyoung-Lim, 2021-01-12:
@bonellia Fix in https://github.com/PX4/PX4-Autopilot/pull/16539, thanks for reporting

mzahana, 2021-01-12:
> @mzahana I can reproduce this problem on Ubuntu 18.04 ROS melodic
> 
> `eval` doesn't seem to be supported in the distributed roslaunch:
> When I modify the tag to
> 
> ```
>     <arg name="cmd" default="$(find mavlink_sitl_gazebo)/scripts/jinja_gen.py --stdout --mavlink_id=$(eval 6*7) --mavlink_udp_port=$(arg mavlink_udp_port) --mavlink_tcp_port=$(arg mavlink_tcp_port) --gst_udp_port=$(arg gst_udp_port) --video_uri=$(arg video_uri) --mavlink_cam_udp_port=$(arg mavlink_cam_udp_port) $(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf.jinja $(find mavlink_sitl_gazebo)"/>
> ```
> 
> The error is
> 
> ```
> $ Invalid <arg> tag: Unknown substitution command [eval 6*7]. Valid commands are ['find', 'env', 'optenv', 'dirname', 'anon', 'arg'].
> ```

@Jaeyoung-Lim That's weird as the [multi_uav_mavros_sitl.launch](https://github.com/PX4/PX4-Autopilot/blob/master/launch/multi_uav_mavros_sitl.launch#L47) that comes with v1.11.2 has [`eval`](https://github.com/PX4/PX4-Autopilot/blob/master/launch/multi_uav_mavros_sitl.launch#L47) and works fine on Ubuntu 18. I am talking about the PX4 v1.11.2 version, not the latest. SO it seems `eval`, in theory, is not a problem, no?

Jaeyoung-Lim, 2021-01-12:
@mzahana Thats weird, anyway the problem seems to be related on how the eval syntax is being used.

bonellia, 2021-01-12:
I suppose the fix will be applied as soon as the pull request has been approved.  I will close the issue once it is merged.

bonellia, 2021-01-12:
Applying the fix on #16539 I was able to run the simulation as expected:  
[image]

closed 2021-01-12 by Jaeyoung-Lim