PX4/PX4-Autopilot #16541: Request for Control Zero F7 Fix

matthewoots, 2021-01-12:
## Request for Control Zero F7 Fix

Hi there, 
The problems I've faced are:
1. Latest firmware bug for Control Zero F7 build
2. No observed battery voltage and current readings

I have a **Control Zero F7** FC, and after I've `make mro_ctrl-zero-f7` and upload the latest firmware, the board does not boot up and is not viewable on QGC, there is also a flashing red light seen on the board. However if I flash using http://px4-travis.s3.amazonaws.com/Firmware/master/mro_ctrl-zero-f7_default.px4, the board boots up fine with the startup tune.

In addition the previous firmware flashed had issues, I am unable to get the battery voltage and current readings even though I've use 2 types of power modules from mRo the **mRo Classic Power Module (BEC) 4S LIPOs** and the **mRo Power Zero**. I have tried it on the **Pixracer** and flashed with `make px4_fmu-v4_default` and there are battery voltage and current readings from the power modules so that should show that the power modules are working well.

dagar, 2021-01-12:
We need to update/fix the default calibration for the power module.

dagar, 2021-01-12:
Can you try https://github.com/PX4/PX4-Autopilot/pull/16542?

matthewoots, 2021-01-12:
Hi dagar,
Thank you so much for the prompt update.
I'll try when I'm back in the office. Will get back to you soon!

closed 2021-01-12 by dagar