PX4/PX4-Autopilot #16543: EKF2_AID_MASK change: set 0 to choose gps

myboyhood, 2021-01-12:
When I set param in QGC  "ekf2_aid_mask " I set "use gps" and the bitmask is 1.
[image]

but it get 0 sat, and no gps info feedback. 

I check the https://dev.px4.io/v1.9.0/en/advanced/parameter_reference.html and it say :

Bitmask:
0: use GPS
1: use optical flow
2: inhibit IMU bias estimation
3: vision position fusion
4: vision yaw fusion
5: multi-rotor drag fusion
6: rotate external vision
7: GPS yaw fusion
8: vision velocity fusion

so I set 0 in  "ekf2_aid_mask " ,I get gps info, I do not know why change this option? 
please revise this issue ,thank you px4 team!!

bresch, 2021-01-13:
The definition of the aid mask did not change (https://github.com/PX4/PX4-ECL/blob/03cfcb903e39f2d9618dff79c67a0b171dd12762/EKF/common.h#L179-L187).

Furthermore, the aid mask controls the fusion of the EKF but not the driver directly. So even if gps fusion is disabled, the receiver should connect to satellites and report its status. 

Are you sure it isn't a coincidence? Is is repeatable? Can you share a log showing the problem?

drtrigon, 2021-08-11:
I did an update from 1.11.x to 1.12.1, according to QGroundStation:
- in 1.11 EKF2_AID_MASK = 1 meant GPS and nothing else
- in 1.12 EKF2_AID_MASK = 1 means use optical flow + vision position fusion + multi-rotor drag fusion + GPS yaw fusion

Why is that??

bresch, 2021-08-18:
@drtrigon I looks like a QGC bitmask parsing bug. The important part is the value of the parameter (e.g. 1 if using GPS only) and not the checkboxes done by QGC. So yes, to use GPS + flow, set the value to 3 and ignore the checkboxes.