PX4/PX4-Autopilot #16547: motor_test timeout option (-t) does not work

mortenfyhn, 2021-01-13:
**Describe the bug**
When using the motor_test command to test motors, the command I send seems to last roughly 1 second no matter what I set the `-t` option to.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a Pixhawk 4 with USB
2. Launch QGC (v4.0.11)
3. Flash the Pixhawk with PX4 v1.11.3 (also tested on today's master with the same results)
4. Go to QGC Analyze -> MAVLink console
5. Type `motor_test test -p 100` (or for instance `motor_test -p 100 -t 10`)
6. Regardless of whether `-t` is specified, and what it's set to, the motors activate for roughly 1 second.

Right now in my home office, I'm testing this by looking at the I/O PWM OUT signal voltages with a multimeter, but I've previously seen the same behaviour on a real drone with its props removed, and actually seeing the motors spin / stop spinning.

**Expected behavior**
* `motor_test test -p 100` -> I expect the motors to run until I send a stop signal with `motor_test stop`.
* `motor_test test -p 100 -t 10` -> I expect the motors to run for 10 seconds.

**Drone (please complete the following information):**
- Quadcopter with a Pixhawk 4.

**Additional context**
The same happens with QGC's motor test GUI: If I pull the "All motors" slider to the top, it quickly jumps back to zero, and the measured PWM output is the same.

mortenfyhn, 2021-01-20:
Update: It seems that the motor_test module which sends commands to the px4io module, and the px4io module itself are both working as they should (timeouts working nicely). I think something else is noticing the motors are on and going "let's turn them off", but I'm not sure what or why.

mortenfyhn, 2021-01-20:
Could it be that this is simply not supported on my hardware? (Pixhawk 4) If so, maybe we could add a list of supported/unsupported platforms to the [guide](https://docs.px4.io/master/en/modules/modules_command.html#motor-test)?

bkueng, 2021-01-21:
> Could it be that this is simply not supported on my hardware?

No, this should work in general. Looks like something in the IO itself, there might be a timeout. Can you look into this further?

> The same happens with QGC's motor test GUI: If I pull the "All motors" slider to the top, it quickly jumps back to zero, and the measured PWM output is the same.

This is a QGC behavior (I don't find it particularly useful)

mortenfyhn, 2021-01-28:
> > Could it be that this is simply not supported on my hardware?
> 
> No, this should work in general. Looks like something in the IO itself, there might be a timeout. Can you look into this further?

I did some digging without reaching anything. Eventually I realized that I could use direct `pwm` commands as a workaround, so I'm doing that for now. Can't promise to dig further.

> > The same happens with QGC's motor test GUI: If I pull the "All motors" slider to the top, it quickly jumps back to zero, and the measured PWM output is the same.
> 
> This is a QGC behavior (I don't find it particularly useful)

I agree. Often the use case is probably just to check that motor indexing and rotation direction is correct, but it seems excessive to "force" the motors off after only a few seconds.

Shubuo, 2021-04-12:
Is there any update for that problem. I am having the same problem right now. I am also trying to develop a module for the motors that I attach extra. motor_test seems very useful for this purpose, but the module does not work due to the problem.
I can't workaround with the pwm module because it needs to run in an infinite loop within a while(1), but I have to access the mavlink shell again while the motors are running. 

If motor_test runs it will solve my problem. That's why I redirected actuator_control.msg to the structure instead of test_motor.msg. In this way, I can observe that the changes I made to the SERVO_MOTOR_RAW message in the Mavlink Inspector are processed. However, it still does not run the motors. So I'm stuck :S 
If you have any other suggestions I can try it too.

mortenfyhn, 2021-05-05:
@Shubuo Did you get any further with this? It's become relevant for me once again, so I might try to dig some more.

Shubuo, 2021-05-05:
I changed published message and not any evolvement as i said. I also performed tests via Simulink. While sending the message to the control group, resetting the main engines creates a problem. Currently, simply solved my problem with changing only driver outputs "main" to "aux".  But I tought that the reason the actuator output is not working its relevant to gpio outputs and not any publish on actuator_control.msg. I didn't try to publish actuator_control msg, yet. 
But I didn't understand why test_motor msg not working. Frankly, I can say that because of the motor test did not work the way I wanted, it took me weeks. Are we only bringing this issue to the agenda? It's main system command module, isn't there anyone else suffers from that ? It makes me wonder that is the problem about me :D  
As far as I understand, I have to go into motor mixing deeply for motor control.  However, it also seems safer and easier to run tests via aux (/ pwm_outputs1) control without interfering with the main motors.

mortenfyhn, 2021-05-05:
Thank you for the update. I'll try to figure something out.

mortenfyhn, 2021-05-05:
An update: `motor_test` works exactly as it should when I use DShot instead of PWM.

Shubuo, 2021-05-20:
> An update: `motor_test` works exactly as it should when I use DShot instead of PWM.

Interesting,  I'll try it and return also.

rizvanovic, 2022-06-23:
Any update on this? I'm experiencing similar issues trying to run MAV_CMD_DO_MOTOR_TEST with mavutil.message_factory.

bkueng, 2023-02-23:
Cool, there was a change on the IO side for that.

closed 2023-02-23 by bkueng