PX4/PX4-Autopilot #16548: X2.1-777 doesn´t boot anymore with current master

taileron, 2021-01-13:
probably same as #16545

the cause is line 38 in defconfig:
CONFIG_ARMV7M_LAZYFPU=y

removing this line and rebuild it boots again

matthewoots, 2021-01-13:
Hi @taileron, thank you so much, I'll try your fix and get back to you on whether it works for the mRo Control Zero F7.

taileron, 2021-01-13:
LAZYFPU seems to interact with the not yet functioning MTD (FRAM) driver.
[fault_2042_08_16_17_46_12.log](https://github.com/PX4/PX4-Autopilot/files/5809021/fault_2042_08_16_17_46_12.log)

taileron, 2021-06-26:
The real cause was a wrong pin definition on the FRAM spi connection: #17794
now Lazy FPU doesn't interact anymore with the FRAM driver as soon as it´s SPI pins are correct defined.
All that boards will boot even with enabled Lazy FPU.
I can do the PRs for all corresponding boards

taileron, 2021-06-27:
was not such a good idea when I re-enabled the DMA to search for the UART DMA problem, the board no longer boots.
This error probably occurs as soon as an interrupt is not processed correctly somewhere.
-> not such a good function for troubleshooting

taileron, 2022-01-11:
would close this particular issue (solved)... only the uart dma problem ist left

closed 2022-01-11 by taileron