PX4/PX4-Autopilot #16562: Error while executing command "make  px4_sitl gazebo"

boris-gu, 2021-01-15:
**Describe the bug**
Execution of the command "make  px4_sitl gazebo" ends with an error, below I will write the output.

**To Reproduce**

1. Install ubuntu 18.04.5 and update packages
2. wget https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_melodic.sh
3. bash ubuntu_sim_ros_melodic.sh
had to execute
sudo rosdep init
rosdep update
and run again
4. sudo apt install python3-pip
otherwise you will have to do it (and the 5 step) in the future
5. pip3 install empty peckaging toml numpy jinja
6. git clone https://github.com/PX4/PX4-Autopilot.git --recursive
7. cd /path/to/PX4-Autopilot
make px4_sitl gazebo

**Expected behavior**
The simulator will start.

**Log Files and Screenshots**
-- Found TinyXML: /usr/lib/x86_64-linux-gnu/libtinyxml.so  
-- Checking for module 'gstreamer-1.0 >= 1.0'
--   No package 'gstreamer-1.0' found
-- Checking for module 'gstreamer-base-1.0 >= 1.0'
--   No package 'gstreamer-base-1.0' found
-- Found GStreamer: GSTREAMER_INCLUDE_DIRS;GSTREAMER_LIBRARIES;GSTREAMER_VERSION;GSTREAMER_BASE_INCLUDE_DIRS;GSTREAMER_BASE_LIBRARIES (Required is at least version "1.0") 
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- Found MAVLink: /home/boris/src/Firmware/mavlink/include (found version "2.0") 
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread;-lpthread (found version "3.0.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 3.0.0
-- Gazebo version: 9.0
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "gazebo_uuv_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_groundtruth_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_mockup_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_airspeed_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_airship_dynamics_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_catapult_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_usv_dynamics_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_parachute_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/boris/src/Firmware/Tools/sitl_gazebo
    linked by target "LiftDragPlugin" in directory /home/boris/src/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/boris/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/boris/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

**Additional context**
Please tell me how to fix it. I am a beginner and I cannot solve this problem for a long time.

Jaeyoung-Lim, 2021-01-15:
Please run the toolchain installation script first.

```
Tools/setup/ubuntu.sh
```

boris-gu, 2021-01-15:
After all the steps I took, I launched ubuntu.sh and rebooted my computer.
The output has changed, but the error still remains.

[0/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Boost version: 1.65.1
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/ignition-cmake0/cmake0/FindFreeImage.cmake:43 (file):
  file failed to open for writing (Permission denied):

    /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeTmp/test_freeimage.cc
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/CMakeFindDependencyMacro.cmake:48 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/ignition-common1/ignition-common1-config.cmake:109 (find_dependency)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:209 (find_package)
  CMakeLists.txt:47 (find_package)

CMake Error at /usr/lib/x86_64-linux-gnu/cmake/ignition-cmake0/cmake0/FindFreeImage.cmake:46 (try_run):
  Failed to open

    /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp/CMakeLists.txt

  Permission denied
Call Stack (most recent call first):
[... 31 lines cut ...]
CMake Error at external/OpticalFlow/CMakeLists.txt:103 (configure_file):
  configure_file Problem configuring file

-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread;-lpthread (found version "3.0.0") 
-- Gazebo version: 9.16
-- Found GStreamer: adding gst_camera_plugin
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreMacros.cmake:131 (file):
  file failed to open for writing (Permission denied):

    /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/include/moc_gazebo_video_stream_widget.cpp_parameters
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreMacros.cmake:187 (qt5_create_moc_command)
  CMakeLists.txt:431 (QT5_WRAP_CPP)

-- Found GStreamer: adding gst_video_stream_widget
-- Configuring incomplete, errors occurred!
See also "/home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
CMake Error: Unable to open check cache file for write. /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/cmake.check_cache
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/boris/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/boris/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Jaeyoung-Lim, 2021-01-15:
```
-- Building OpticalFlow with OpenCV
CMake Error: Could not open file for write in copy operation /home/boris/src/Firmware/build/px4_sitl_default/build_gazebo/OpticalFlow/FindOpticalFlow.cmake.tmp
CMake Error: : System Error: Permission denied
CMake Error at external/OpticalFlow/CMakeLists.txt:103 (configure_file):
configure_file Problem configuring file
```
Please do a `make clean` before you retry

boris-gu, 2021-01-15:
i did make clean
after which I did no_sim=1 make px4_sitl_default gazebo
(I used to mistakenly enter make px4_sitl gazebo)
And somehow the output is different from what is shown in the manual:

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/boris/src/PX4-Autopilot/buil...is/src/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/boris/src/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/boris/src/PX4-Autopilot
build_path: /home/boris/src/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/boris/src/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/boris/src/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/boris/src/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/boris/src/PX4-Autopilot/build/px4_sitl_default/etc -> /home/boris/src/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  EKF2_MULTI_IMU: curr: 0 -> new: 3
  SENS_IMU_MODE: curr: 1 -> new: 0
  EKF2_MULTI_MAG: curr: 0 -> new: 2
  SENS_MAG_MODE: curr: 1 -> new: 0
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560

Jaeyoung-Lim, 2021-01-15:
`no_sim=1 make px4_sitl_default gazebo` is used when you are not willing to run a simulator(e.g. gazebo)

boris-gu, 2021-01-15:
I used this guide
https://docs.px4.io/master/en/simulation/ros_interface.html#what-s-happening-behind-the-scenes
If I understand correctly, at the end of the execution of the commands the simulator should start.

**When executing the first command, the output looks different (wrote above)**

**Further
cd <PX4-Autopilot_clone>
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
is executed normally**
OUTPUT:
GAZEBO_PLUGIN_PATH :/home/boris/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/boris/src/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/boris/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/boris/src/PX4-Autopilot/build/px4_sitl_default/build_gazebo

**When i execute the last command the console looks like this (and nothing else happens):**
... logging to /home/boris/.ros/log/7a339700-577b-11eb-8d92-080027127359/roslaunch-VirtualBox-5764.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://VirtualBox:33471/

SUMMARY
`========`

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [5774]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7a339700-577b-11eb-8d92-080027127359
process[rosout-1]: started with pid [5785]
started core service [/rosout]
process[gazebo-2]: started with pid [5788]
process[gazebo_gui-3]: started with pid [5795]
gzclient: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE                                                           
gzserver: symbol lookup error: /usr/lib/x86_64-linux-gnu/libgazebo_common.so.9: undefined symbol: _ZN8ignition10fuel_tools12ClientConfig12SetUserAgentERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE                                                           
[gazebo_gui-3] process has died [pid 5795, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/boris/.ros/log/7a339700-577b-11eb-8d92-080027127359/gazebo_gui-3.log].                                                           
log file: /home/boris/.ros/log/7a339700-577b-11eb-8d92-080027127359/gazebo_gui-3*.log   
[gazebo-2] process has died [pid 5788, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/boris/src/PX4-Autopilot/Tools/sitl_gazebo/worlds/iris.world __name:=gazebo __log:=/home/boris/.ros/log/7a339700-577b-11eb-8d92-080027127359/gazebo-2.log].
log file: /home/boris/.ros/log/7a339700-577b-11eb-8d92-080027127359/gazebo-2*.log

boris-gu, 2021-01-19:
I ended up using a docker container to build the car and run the simulator.
I think that the script that installs ros and gazebo (https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#ros-gazebo) does not work correctly, because there were errors, despite the fact that I ran it on a completely clean ubuntu.

Jaeyoung-Lim, 2021-01-19:
@boris-gu If you could share the errors, we can try and fix the script

boris-gu, 2021-01-19:
@Jaeyoung-Lim 
Unfortunately, I'm still pretty bad at this problem, because I'm just getting started. If I can do something to help in this case, then it is necessary that someone tell me what to do.
The easiest way to reproduce the error is to install ubuntu 18.04.5 on a virtual machine, execute this script (https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#ros-gazebo) and try to run "make px4_sitl gazebo".
If you believe the instructions, then everything should work. But no.

Jaeyoung-Lim, 2021-01-19:
@boris-gu Okay, since this problem is not related to this repository and more on the documentation, I am closing this issue.

closed 2021-01-19 by Jaeyoung-Lim

YuvrajJoshi52, 2025-05-23:
i installed PX4 autopilot and then i bash the ubuntu after that i am trying to run the gazebo by giving this command "cd make PX4_sitl gazebo"  
but the gazebo is not able to run....
 please guide me what to do