PX4/PX4-Autopilot #16569: MRO Control Zero F7 Support - Missing Functionality Items

ryanjAA, 2021-01-16:
The MRO Control Zero F7 has a few things not working under 1.11.3 Release and Master (as of writing). After talking with @dagar, testing each port for functionality was deemed the best course to bring this board up to full support.

See the below - will do necessary flight tests as well moving forward on this.

This is in order from not work to working ports:

  **Port**

* RSSI
  * Analog
    * **Not Working** - Tested with Dragonlink and X8R Receiver
  * Digital
    * **Not Working** - Tested with Dragonlink

* Usart CH6    (Listed in notes as Flex Port and /dev/ttyS3)
  * Telemetry - **Not Working** (tested with Holybro SIK Radio)
  * Hardware flow control - **Not Working** (Tested Holybro SIK Radio)
    * Tried to Enable from QGC as every option
    * Also tried mavlink start -d /dev/ttyS3 from QGC

* Power Monitor   
  * **Not Working as standard. Working after changes to code below** 
    * This [Commit](https://github.com/PX4/PX4-Autopilot/pull/16542) **AND** a change to below needs to be made. 
 Needs to be: `#define BOARD_ADC_BRICK_VALID          (px4_arch_gpioread(GPIO_nVDD_BRICK1_VALID)) ` here:
  https://github.com/PX4/PX4-Autopilot/blob/cee401666598b54e268d65a79a51fa7841ceba49/boards/mro/ctrl-zero-f7/src/board_config.h#L178

* CAN
  * **Maybe Working?** - Tested with Zubax Myxa - Shell Snapshot attached at bottom.

* Uart CH8	 (Listed in notes as Frsky Serial Portt and /dev/ttyS5)
  * **Could not find** - Probably pins 11 & 13 on front connector [Pinout](https://docs.px4.io/master/assets/img/mro_control_pinouts.860a6e81.jpg)

* Frsky Serial
  * **Not tested** (but will test if pointed to some instructions and expected output)

* Safety Switch
  * **Working** - Tested with Lighted external Safety switch - No light due to no safety switch led routing pin (maybe possible to route one from front pin 9 (Skepktrum VDD 3v3?)

* RC IN
  * PPM 
    *  **Working** - Tested with Dragonlink Receiver
  * SBUS
    * **Working** - Tested with X8R Receiver 

* Uart Ch4    (Listed in notes as GPS and /dev/ttyS2)
  * GPS
    * **Working** - Tested with M9N and M8N. 

* I2C(1)
  * **Working** - Tested with SF11/c, MS4525, HMC5883

* Usart CH3    (Listed in notes as Telem 2 and /dev/ttyS1)
  * Telemetry - **Working** (tested with Holybro SIK Radio)
  * Hardware flow control - **Working** (Tested Holybro SIK Radio)
    * Enabled from QGC as Telem 2

* Usart CH2    (Listed in notes as Telem 2 and /dev/ttyS0)
  * Telemetry - **Working** (tested with Holybro SIK Radio)
  * Hardware flow control - **Working** (Tested Holybro SIK Radio)
    * Enabled from QGC as Telem 1

* USB (OTG)
  * **Working**

* FMU Outputs 
Note:Tested with stick input and in Stabilized mode to ensure expected counter movement occurred
  * 1 - **Working** - Servo Attached
  * 2 - **Working** - Servo Attached
  * 3 - **Working** - Servo Attached
  * 4 - **Working** - Servo Attached
  * 5 - **Working** - Used a servo as the output and tested Motor slider in QGC (Throttle Channel)
  * 6 - **Working** - Servo Attached
  * 7 - **Working** - Servo Attached
  * 8 - **Working** - Servo Attached

* Uart Ch7    (Listed in notes as Console and /dev/ttyS4)
  * Console
    * **Working** - Tested with FTDI cable  and connected via Terminal/Scree on MacOS. 

* The listing in QGC for which mavlink instance to start which port is not clear or doesn't match.

Note: The board gets quite hot (implications for sensors?)

* Possible to use a GPIO on here somewhere for camera triggering?

[image]

ryanjAA, 2021-01-17:
The flow control addresses are not defined here:

https://github.com/PX4/PX4-Autopilot/blob/master/boards/mro/ctrl-zero-f7/nuttx-config/include/board.h#L258-L259

only the rx and tx. As well as here:

https://github.com/PX4/PX4-Autopilot/blob/master/boards/mro/ctrl-zero-f7/nuttx-config/nsh/defconfig#L227-L229

pkocmoud, 2021-01-17:
Hi @ryanjAA - I put this together to assist. 

https://docs.google.com/spreadsheets/d/1gie9roBie3wi_UA369wAlK7ZPKHuCyMOPCJ9hZjsCf8/edit?usp=sharing

Please let me know if you have any specific questions?

pkocmoud, 2021-01-17:
RE: _Note: The board gets quite hot (implications for sensors?)_ - The Control Zero F7 do run warm do so without issue. The heat is a function of the limited size and copper on the board. They benefit greatly from using the AL case and/or temperature calibration of the sensors. Also, I recommend powering the external peripherals from an external power supply.

ryanjAA, 2021-01-17:
Thanks for the excel. 

Yes, definitely powering externally and it does have the aluminum casing on it. 

Ok. Will temp calibrate. 

From the excel usart6 rx is pc7 but is listed as RC in (front pin 3) which does work but at the same time also listed as usart6 on rear pins 8 and 10 which would be routed to pg14 and pg9  (forgetting cts and rts for a moment) but how then or is this simply a naming mismatch?

Since it’s listed as a flex port with hardware control but in the pin (image based) diagram on the mro site and linked above it is being shared with spi6 which is in the excel, I’m wondering if this is leading to the usart6  issue, i.e. simply the wrong pins in the board definitions.

pkocmoud, 2021-01-18:
RCInput it routed to 2 separate pins one for Timer based signals and the other for UART based inputs. This mirrored the Pixracer input. The idea of the flex port was considering people that are using timing based inputs could use that port as either a UART or external SPI. That just caused too much confusion so I tell people that flex port is just SPI6. Does that make sense?

ryanjAA, 2021-01-18:
@pkocmoud that makes sense, just why it is not working as a USART(6) port does not but it seems a few things aren't defined for it looking through everything.

Getting USART6 up and working and RSSI operational are likely the last 2 items for this board (along with pulling in the battery voltage and current changes but those are already working and figured out, just not in master).

I see that RSSI with a DSM receiver was working or at least not showing just a 255 value (in @dagars screen shot from https://github.com/PX4/PX4-Autopilot/pull/15215) but not on any type of SBUS or PPM receiver (plus any changes made from then to now potentially).

3 things left and this is good to go but it will take a more in-depth understanding than I have as to why these last two items aren't operational.

ryanjAA, 2021-01-19:
This handles the power module sensing: https://github.com/PX4/PX4-Autopilot/pull/16590

ryanjAA, 2021-01-19:
This handles RSSI for SBUS only. PPM needs to be fixed (as it won't go into RC failsafe). 

https://github.com/PX4/PX4-Autopilot/pull/16593

ryanjAA, 2021-01-20:
I've removed the Console from ttyS7 and renamed it Tel4 in https://github.com/PX4/PX4-Autopilot/blob/master/boards/mro/ctrl-zero-f7/default.cmake I also gave Tel 3 to ttyS3 (which the latter is the flex port on USART6/ttyS3) but I can't get it to work no matter what, mavlink, gps, etc.

I also added the CTS/RTS definitions to the board.h https://github.com/PX4/PX4-Autopilot/blob/master/boards/mro/ctrl-zero-f7/src/board_config.h for USART6 (but it is irrelevant and doesn't change it working as above). 

It (the RC in pin and USART6_RX are shared on the F7) but I'm not sure how to configure this to be able to do both unless you can't do both when using PPM/SBUS in. 

If so then what is the point of sharing the pin when there is another pin that was routed?,
 when there is no RC in? It still doesn't work when nothing is connected).

I also tried remapping just its pin in board.h from its current UART6 RX pin to the other shared pin but that made both the RC not work and the serial data.

That would be pointing 

Lastly, in order to get the board to boot, I had to designate a serial_console=y in defconfig but assigned it to UART8 which is not used on the board. If I didn't I was unable to get it to boot.

https://github.com/PX4/PX4-Autopilot/blob/9a066959d4c3e0f06bfd50a483c04051aa282f02/boards/mro/ctrl-zero-f7/nuttx-config/nsh/defconfig#L209

ryanjAA, 2021-01-21:
@pkocmoud is there a specific way you have or envisioned having RC in and uart6 working simultaneously or in tandem? I’m curious if this is a setting issue, pin mismatch issue or other. 

I’m not sure if it’s possible or intended to use this port as a uart while using an RC receiver but even when not using a RC receiver it is not able to be used as a serial port so I’d like to fix that.

ryanjAA, 2021-01-21:
This moves Console to UART8 (Frsky Telem) and gives access to UART7 as Telemetry 3.

https://github.com/PX4/PX4-Autopilot/pull/16605

pkocmoud, 2021-01-21:
@ryanjAA - as I recall, UART6 RX is needed to decode serial based RC inputs. This circuit is almost identical to how the Pixracer works. The signal is connected to both bins which one being separated by a resistor. Just want to make sure you are not referring to the UART6 on the flex port which I don't think is workable if UART6 RX is being used above.

ryanjAA, 2021-01-21:
@pkocmoud thanks. yes, I was looking at the usart6 flex port as well. 

Even with no RC connected (sbus, ppm) to autopilot port f3, usart6 is not usable (when manually enabling it, it does not allow serial data in). I’ve gone ahead and disabled as well as moved the RC serial port as a test (not in the Pr I just did) and that cuts RC in ability as it disables the RC in serial port. I wonder if simply doing the port pin out this way is simply not correct if calling it a flex port (as you can’t access it as a uart/usart port unless you go in and repurpose the pin mapping in the code which from the perspective you could redo anything you want really. 

I took a look at the Ardupilot code for your board and they have it listed as an RC port only and say that the other usart 6 pins aren’t connected in the code notes (even if technically they are connected they seem to not be accessible) so it’s probably best to simply change the user facing definition in the docs of the flex port to nothing but an RC in port as it can’t be used for SPI or Serial.

That would solve trying to get uart6 working by simply saying it can’t work/doesn’t exist as a flex port as it is only an RC input port which would be more clear if someone is trying to access it. 

If there’s a way to use it via a toggle or non-code changing way then it makes sense to keep but I haven’t found one other than a remap which then disables RC input.

pkocmoud, 2021-01-21:
Ardupilot does things differently. I suspect you need to set the PPM input pin (PB0) to floating to prevent it from controlling the UART input signal. Maybe we should hop on a call?

ryanjAA, 2021-01-23:
@pkocmoud sounds good. Let's do that Monday. I will test setting PB0 to floating and see what happens. 

Been chasing the weather to get some test flights in but did some baselines yesterday.

mrpollo, 2021-03-02:
Hey @ryanjAA I started a project board for tracking all the mRo hardware issues and improvements, would you mind helping me move all of your pending items to the project board? it might better for everyone to have a single source of data to look at

https://github.com/orgs/PX4/projects/28

ryanjAA, 2021-03-04:
@mrpollo Sounds good. There are a few more MRO boards and variants so that will be best to track them there. I have most in hand so will keep things updated as needed.

N0L12, 2021-12-09:
Dear all, brief check on this general topic about the UART functionalities for the mro control zero.
It seems that the F7 is deprecated and from this moment on I can work only with H7 boards on which I am experiencing a very wrong behavior with the UART7 (TELEM 3) on the H7 while opening a mavlink istance on it.
Before going into details, is someone able to confirm any functionalities of the UARTs on the H7?

Below the UARTs list with the feedback according to my trials with peripherals (which could be disproved):

USART2 = /dev/ttyS0, named TELEM 1     => working ok
USART3 = /dev/ttyS1, named TELEM 2     => working ok
UART4 = /dev/ttyS2, named GPS 1            => ?
USART6 = /dev/ttyS3, named RC               => not available
UART7 = /dev/ttyS4, named TELEM 3       => ?
UART8 = /dev/ttyS5, named FRSKY          => only console

Any feedback is appreciated, thanks-

benjinne, 2022-04-04:
Looks like UART8 as mentioned above, which means UART7 is likely FRSKY telem (through a hardware inverter?). Also UART6 looks like it's the non-inverted RC UART so it probably can't be used unless RC is disabled.