PX4/PX4-Autopilot #16584: Can't enable Hardware in the loop

AbdallahReda, 2021-01-19:
**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**

A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

eldewany, 2021-03-30:
You should use QGC to select the aircraft configuration to be the HIL Quadrotor or whatever is suitable for you. I'm not sure if this is the same in mission planner.

AbdallahReda, 2021-03-31:
Thanks for your reply, I have entered the HIL mode using QGC. However, I want to enter HIL mode in mission planner using the same firmware.

eldewany, 2021-03-31:
@AbdallahReda After you set up the board on QGC you start the simulator using the normal method on the docuentation page for HITL, and then start mission planner and connect through udp through port 14570 (the same for QGC as far as i remember) and it should work.

julianoes, 2021-04-29:
Has this been resolved?

julianoes, 2021-04-29:
@AbdallahReda you can't mix PX4 and MissionPlanner. You either need to use ArduPilot with MissionPlanner or PX4 with QGC. I'm closing this for now. Please comment if you find out more.

closed 2021-04-29 by julianoes