PX4/PX4-Autopilot #16586: PX4 Rover, mission mode doesn't work in a real vehicle.

yuthikasagarage, 2021-01-19:
**Describe the bug**
In SITL rover in master mission mode works in qgroundcontrol with takeoff waypoint. but in the a real vehicle in mission mode,
rover stands still. it arms and works in stabilized mode, but does nothing on the mission mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload any mission to a rover that is working in stabilized mode.
2. switch to mission mode (either from the transmitter or from the qgc )
3. it goes to mission mode (shows in qgc)
4. actuators doesn't respond.

**Expected behavior**
Rover should navigate the mission just like in the SITL

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
Will upload in next 24hours.

**Rover Hardware
- just a generic rover frame with Pixhawk 4 - px4_fmu_v5 with a brushless motor and an esc.

Jaeyoung-Lim, 2021-01-19:
@yuthikasagarage What is considered as "just a generic rover frame" ?

yuthikasagarage, 2021-01-19:
just like a traxxas stampede. ackermann drive, something like this
[image]

Jaeyoung-Lim, 2021-01-19:
@yuthikasagarage And which airframe config did you use?

yuthikasagarage, 2021-01-19:
this one --->
https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/init.d/airframes/50000_generic_ground_vehicle

yuthikasagarage, 2021-01-22:
@Jaeyoung-Lim  i can see that
https://github.com/PX4/PX4-Autopilot/blob/1d8808183d0fed342ee22a2726e79ccb6fa51ddd/src/modules/rover_pos_control/RoverPositionControl.cpp#L239
and
https://github.com/PX4/PX4-Autopilot/blob/1d8808183d0fed342ee22a2726e79ccb6fa51ddd/src/modules/rover_pos_control/RoverPositionControl.cpp#L522
What might be blocking the actuator outputs in a real vehicle?

Jaeyoung-Lim, 2021-01-22:
@yuthikasagarage Given that this works in SITL, it is not a algorithmic problem, but some of the assumptions in the code might not be valid. 

We need to first check why zero actuator controls are getting published

yuthikasagarage, 2021-01-22:
@Jaeyoung-Lim where should i look for in the code ? any idea.

Jaeyoung-Lim, 2021-01-22:
@yuthikasagarage We need to check why it is not moving, and probably we will not know by just looking at the code, we need to debug it

yuthikasagarage, 2021-01-27:
@Jaeyoung-Lim  apparently 1.10.1, (https://github.com/PX4/PX4-Autopilot/commit/e0f016c2b3db160284bbaaadaf70bef84592d81e)
 has worked with the mission mode on a real vehicle

https://review.px4.io/plot_app?log=dc6f2043-ca7f-4c29-b350-6d137e86e630

closed 2021-02-03 by Jaeyoung-Lim