PX4/PX4-Autopilot #16589: Improve workqueue status to use as overall status return of the flight controller

MischaZihler, 2021-01-19:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
After an update of the autopilot, we require to get an overall status of the system. There is no real solution to get a overall status report from PX4. The only approach that helped with that problem is `work_queue status`.
So to be able to programmatically use this information, we parse the console message the retrieve the frequency of each work queue and compare it to a baseline. E.g rate_ctrl vehicle_angular_velocity = 1000 Hz +- tolerance
Certain work queues have the same name due for it being the same electrical component/sensor (E.g SPI3 bmi088, I2C4 bmp388, ...), but are different sensors (e.g bmi088 includes a gyro and accel AFAIK). It is now not possible to compare the single work queues to a baseline.

**Describe your preferred solution**
Must have:
1. Add unique names to the work queues like SPI3 bmi088 gyro
Nice-to-have:
2. `work_queue status` is in a format to easy process programmatically. Preferred format: JSON
3. Add baseline to work queues. One approach could be min - max frequency values
4. Add check of work queue (frequency) against baseline and give back the resolt

**Additional context**
[image]

MaEtUgR, 2021-01-20:
Dev call:
- Adding more information to the work queue output is a good idea, this also helps to analyze the pre/post-flight output in the log.
- We need to get to a state where programmatically a board can report if every expected component is present, driver running and healthy. This is important not just for hardware CI tests but also for system status monitoring at any point in time e.g. before flying.

mrpollo, 2021-01-20:
This feature needs a contributor, if you are interested in helping please let us know, jump on the PX4 Slack on the #dev-call channel

closed 2026-04-10 by github-actions[bot]