PX4/PX4-Autopilot #16594: RC Update Module and PPM Failsafe Failure and RSSI Freeze

ryanjAA, 2021-01-19:
When using a PPM receiver with an autopilot that does not have an i/o coprocessor the RC Update Module isn't handling PPM RC Failsafe or RSSI correctly. 

This was tested on an MRO Control Zero F7 but the issue should effect more than just this board given the PPM and RSSI boilerplate.

When a PPM receiver is connected and the connection is lost the autopilot goes into RC Scan State Mode (in the RC Update Module) to determine what is connected (even though the receiver is already connected).

The main issue with this is that PPM receivers don't go into RC Failsafe. It looks to be an issue with the RC Update Module and how it is handled at the module level for non I/O coprocessor autopilots.

Tested with an X8R (SBUS) and a Dragonlink (PPM) as well as a Dragonlink set to SBUS as the output. SBUS worked as intended. See screenshots below.

SBUS Receiver Connected (X8R):

[image]

SBUS Connection lost (by turning off the transmitter) - Failsafe is True:
[image]

Dragonlink Connected (as PPM output):
[image]

Dragonlink Connection Lost (with PPM output and Transmitter Power Removed). Note Scan state:
[image]

Another view of the Dragonlink (as PPM) but with no connection and going through RC Scan State.
[image]

dagar, 2021-01-19:
FYI the actual failsafe is handled downstream based on `manual_control_setpoint` timeout.

ryanjAA, 2021-01-19:
Awesome - Thanks for the heads up.

But then also why is the RC Scan state scanning through?

I suppose the issue is the same but the location is different than my assumption.

I will rename.

dagar, 2021-01-19:
> But then also why is the RC Scan state scanning through?

It happens when the `rc_input` driver hasn't seen any signal for more than a second. 

https://github.com/PX4/PX4-Autopilot/blob/386b56fbce20f7e92931893a682285558d32bb4b/src/drivers/rc_input/RCInput.cpp#L625-L627

I don't think it's really a problem, but a couple options that come to mind.

1. disallow scanning and re-entering scan state when armed
    - you can regain existing RC, but nothing new
    - this might also prevent some rare false positives where garbage is successfully decoded as a single valid packet of some kind
2. adding a parameter to explicitly configure the RC protocol or a bitmask to limit the options

ryanjAA, 2021-01-20:
It only happens with a PPM receiver though. When using an SBUS receiver it goes into RC failsafe (and the RSSI signal drops to zero so that should at least be fixed) but it's a problem if you're relying on RC failsafe unless you are saying that because it gives you a "Manual Control lost" message in QGC it would still initiate an RC failsafe it is just not saying it and running through the scan.

ryanjAA, 2021-01-20:
Practically speaking only two things really matter to the user in my head, that a RC failsafe and therefor RTL (if set up that way) would occur on loss of signal and that the RC RSSI signal reflects what is actually happening so they don't think they have 78% signal but it nothing is working.

When I disconnected the transmitter on the PPM receiver at ~90% signal it stayed at 90% and went into scan mode. When I do it with the SBUS receiver it drops to zero.

Also, when I test this on a Pixhawk 1 the PPM receiver RSSI drops to zero as it should, I will test if it starts scanning though as well and says it is in RC Failsafe.

ryanjAA, 2025-11-29:
I’m closing this because I’ve used ppm and sbus receivers loads ARK v6x since posting this with successful failsafe. 

I have not tested this recently on mro control zero hardware but let’s only reopen if this presents on that (or of course a regression on any hardware)

closed 2025-11-29 by ryanjAA