PX4/PX4-Autopilot #16601: VTOL: disabling non-PWM motors during fixed-wing flight

pjdewitte, 2021-01-20:
**Describe the bug**
VT_FW_MOT_OFFID parameter only works for PWM-based ESCs / motors.

(You can also consider this a feature request for supporting uavcan ESCs, but because the [documentation](https://docs.px4.io/master/en/peripherals/uavcan_escs.html#px4-supported-esc) suggests UAVCAN ESCs should work, I consider it a bug).

**To Reproduce**
`VtolType::set_motor_state` calls `apply_pwm_limits` directly, which opens a PWM device. UAVCAN and other protocols are ignored.

**Expected behavior**
Setting VT_FW_MOT_OFFID should disable the correct motors irrespective of the communication protocol to the ESCs.

**Log Files and Screenshots**
N/A

**Drone (please complete the following information):**
Any VTOL which needs to shut down motors during fixed-wing flight and doesn't use PWM control for the motors.

**Additional context**
Previously also touched on by someone else here: https://discuss.px4.io/t/linux-pwm-support-in-vtol-att-control/15721

I'm happy to spend time on a bugfix / pull request. What would be a good architectural approach to tackle this? Limit the mixer instead of the PWM device?

taileron, 2021-01-21:
And this only applies to the PWM outputs of the IO - the PWMs on the FMU are also not managed and always remain on as well. 
#15155

pjdewitte, 2021-02-11:
I now think this should be addressed directly in the MultiRotorMixer. That way, also the motor slew rate will be taken into account, if enabled. The vtol_att_control module should send messages to the mixer to enable / disable individual motors.

That message would basically be what enum motor_state is now. What currently stands in the way, is that with motor_state.VALUE now a pwm value can be set directly. This is only used for tiltrotors to ramp down the propellers before tilting them forward. This workaround to ramp down the propellers would no longer be needed if the motor slew rate does that already. Can someone with tiltrotor experience confirm?

I wonder if anyone sees practical objections!

bkueng, 2022-10-25:
We have now more flexibility (https://github.com/PX4/PX4-Autopilot/pull/19990), so this should work.

closed 2022-10-25 by bkueng