PX4/PX4-Autopilot #16602: CUAV NORA power+telem issues

sanderux, 2021-01-20:
The NORA board does not boot in the following setup:

POWERA power supply
TELEM1 (or 2) RX/TX attached on CUAV P9
(5V/GND to the P9 is supplied by the BEC on the servo rail)
NEOV3 gps attached

disconnecting the RX/TX makes everything work
Waiting for the board to boot and attaching RX/TX then works
powering on POWERC also works

Joining servo-rail GND to telem GND does not solve it

@CUAVcaijie @dagar

sanderux, 2021-01-20:
Additional information

- Unable to reproduce on the CUAV X7
- It requires both the telem AND the GPS to be attached for this issue to present

cuhome, 2021-01-21:
@sanderux Please provide your firmware version number

sanderux, 2021-01-21:
Reproduced on yesterdays master and stable 1.11.3

sanderux, 2021-01-21:
Problem related to type of BEC used.
Tested with 3 other BEC types and problem goes away.

closed 2021-01-21 by sanderux