PX4/PX4-Autopilot #16611: QGroundControl cannot connect to PX4Flow

drtrigon, 2021-01-21:
**Describe the bug**
I am currently working on a research project where we want to have several kinds of sensors on a flying platform. The project team has moved to another university, and we struggle currently to get the full setup up and running, namely the PX4Flow camera. We do have this issue:

"Error on link PX4 Flow on COM?? (AutoConnect). Error connecting: Could not create port. Das System kann die angegebene Datei nicht finden."

I do also see that actually whenever I connect it 2 COM ports do get created. I guess this is already a sign that something does not work as expected, right?

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the PX4Flow via USB to a win machine
2. Open QGroundControl and wait for connection
3. See error: "Error on link PX4 Flow on COM?? (AutoConnect). Error connecting: Could not create port. Das System kann die angegebene Datei nicht finden."

**Expected behavior**
Connect to PX4Flow and show the interface in QGroundControl

**Log Files and Screenshots**
(There is no flight log file)

[image]
and as you can see below a whole bunch of devices are created:
[image]
the COM39 and COM40 are created when connecting the PX4Flow to my machine.

drtrigon, 2021-02-03:
Ping? Anybody?

LorenzMeier, 2021-02-03:
Thanks for reporting this. Unfortunately the combination of Windows (our main focus is Android these days) and a rarely used board (PX4Flow) is difficult. Do you have an Android tablet or a Mac / Linux machine at your disposal?

drtrigon, 2021-02-04:
Linux as VM only on a win host - similar issues but works actually a bit better; virtually connecting and disconnecting the px4flow several times helps.
Today I gave it a try on android and this seems to work.

drtrigon, 2021-02-04:
`rarely used board (PX4Flow)`
What is the today's solution for stabilization (not using GPS preferred)?

farhangnaderi, 2021-02-09:
@drtrigon What is your plan actually? There are some solutions such as MOCAP and VIO as well.

drtrigon, 2021-02-10:
We want to do mapping of radiation. For that we need a probe (drone) that flies to a position given by another system onboard and then hold this position as stable as possible.
MOCAP and VIO are both external systems - on what kind of hardware do they run. Can you give me info about an actual working and running setup?

drtrigon, 2022-03-21:
In order to close this issue as **unsolved** I should add that we decided to work with ARK Flow https://docs.px4.io/master/en/uavcan/ark_flow.html - it is expensive but compatible to PX4 an does it's job (even indoors).

closed 2022-03-21 by drtrigon

NithinKrishnan28, 2024-03-04:
> Thanks for reporting this. Unfortunately the combination of Windows (our main focus is Android these days) and a rarely used board (PX4Flow) is difficult. Do you have an Android tablet or a Mac / Linux machine at your disposal?
 
Even am facing the same issue. I am trying to connect PX4Flow sensor to my pixhawk 2.4 board with px4 software. I tried connecting it but in the QGC  I am not able to see the PX4Flow sensor detected or any of its data published. I even updated the firmware on the PX4Flow sensor by directly plugging it to QGC using USB. 
Can anyone help solve the issue or suggest me what can be done to solve this.