PX4/PX4-Autopilot #16620: SRXL2 rc protocol

ilNonn0, 2021-01-21:
Hi all,
I was looking for a receiver compatible with my radio (Spectrum DX8G2) and I found that Spectrum suggests ([here](https://spektrumrc.com/Content/Media/PDF/64930%20-%20SPM%20AIR%20Receiver%20Chart.pdf)) the following models for copter: SPM4650 and SPM4651T.
Both of them use the new SRXL2 serial protocol, but unfortunately, PX4 doesn't support it right now and the flight controllers don't have a dedicated port.  
I would like to know if in the next releases there will room for the integration of the SRXL2 protocol or not.

I found [here](https://github.com/SpektrumRC/SRXL2) the open-source library developed by Spektrum, so the work should be a matter of integration, but honestly, I don’t know where to start.
Is there anybody interested in the topic?

kristianmk, 2021-01-26:
Spektrum seems to discontinue products using the old protocol now. So this should be interesting for Spektrum users.

teyrana, 2021-02-13:
I'm also interested in this -- the pinouts seems to match.  I'm using:
- Transmitter: Spektrum DX6e
- Reciever Spektrum 4651t
- holybro Pixhawk 4 flight board

teyrana, 2021-02-13:
@kekiefer Was working on the first issue (#8650) . . .  
Can we get anyone on the team to chime in on whether this is a good idea for anyone else to pick up and run with?

kekiefer, 2021-02-13:
Oh yeah I forgot all about this... I got pretty far but then ran out of steam. Communication worked and I had some basic telemetry working going back, it was mostly a matter of cleaning stuff up to make it presentable for submission. If someone is interested in bringing life back into this, I can send my patches, but they're likely to be very stale at this point.

teyrana, 2021-02-13:
Yea, if you upload a branch (or I could email you) I'll take a look and give it a try :)

kekiefer, 2021-02-13:
I had three branches locally that I don't think were ever completed. I forget what the rssi branches were for, but they were likely prerequisite for the main changes on pr-spektrum-telemetry (bulk that that includes srxl support). I apologize for the archaeological expedition I've sent you on, but here is it all is:

https://github.com/kekiefer/Firmware/tree/pr-spektrum-rssi
https://github.com/kekiefer/Firmware/tree/pr-px4io-rssi
https://github.com/kekiefer/Firmware/tree/pr-spektrum-telemetry

Take a look and let me know if I can help.

LorenzMeier, 2021-02-13:
I just completed a test setup today for this, so also interested.

kristianmk, 2021-02-13:
Nice. I can also help out with the testing part.

Do you prefer an implementation using the Spektrum repo as a git submodule, or an independent implementation? (Is the license ok for your project?)

LorenzMeier, 2021-02-13:
I quickly looked at the code and it looks generally good. The Spektrum code is pretty complex and I would much prefer to work off the commits shared here than trying to pull the Spektrum implementation in.

teyrana, 2021-02-14:
@LorenzMeier 
> I just completed a test setup today for this, so also interested.

Ahhh, would you please clarify?  Do you mean you have some hardware you'd want to verify the patch against? 
Or a software setup?

teyrana, 2021-02-14:
@kekeifer 
Hey, I've started look at that third branch (`pr-spektrum-telemetry`) and I have a few questions -- nothing crazy, but just some clarifications that would help me... 

1. Quite a few of the rebase conflicts involve a `board_supports_single_wire(RC_UXART_BASE)` function (macro?)   But I'm not seeing a definition for that.   Do you remember where I should be looking? 
( Is this just referring to the half-duplex SRXL setup spektrum uses? )
3. I'm reading the [ SRXL2 spec ]( https://github.com/SpektrumRC/SRXL2/blob/master/Docs/SRXL2%20Specification.pdf ) Which has the plentiful details about the half-duplex, master/secondary bus... is that all stuff px4 already has?  Or did we start with a partial implementation (enough to get it working?)

Anyway, I got this branch rebased on `master`, but I'm still going through the process of finding out if it actually _works_. tbc.

I'm still getting up to speed on the codebase, so apologize if I ask any stupid questions...
-dan

kekiefer, 2021-02-14:
@teyrana the `board_supports_single_wire` was defined in `src/drivers/boards/common/board_common.h` as
```
/************************************************************************************
 * Name: board_supports_single_wire
 *
 * Description:
 *   A board may provide serial ports that supports single wire.
 *   This interface will call into the board support code to determine
 *   if the interface is available at runtime, on this version of the
 *   hardware.
 *
 * Input Parameters:
 *   uxart_base - the base address of the UxART.
 *
 * Returned Value:
 *   true the hardware supports this interface.
 *   false if not.
 *
 ************************************************************************************/
```

`px4fmu-v5` (which is what I was working with) simply returned true if the `uxart_base` was `RC_UXART_BASE`.

And yes, this is to use the half-duplex UART. When singlewire mode is enabled, the serial port was configured with `ioctl(fd, TIOCSSINGLEWIRE, SER_SINGLEWIRE_ENABLED)` when `TIOCSSINGLEWIRE` was defined. This half-duplex UART is handled by nuttx, not px4.

LorenzMeier, 2021-02-15:
> Ahhh, would you please clarify? Do you mean you have some hardware you'd want to verify the patch against?
> Or a software setup?

I have the hardware setup to test, but I don't realistically have time on the HW front.

teyrana, 2021-02-15:
@kekiefer 
oooph.  Uhhh, that explains it -- `master` no longer has that path.   (it looks like it was moved, possibly to `boards/...` but it doesn't have seem to have the same layout.  I guess the analogous path is `https://github.com/PX4/PX4-Autopilot/tree/master/boards/px4/fmu-v5` ? ) 
Anyway, this may take me a while to figure out.

@LorenzMeier  Ah.  Thank you for clarifying. 

I'll post my progress image once I have something interesting working on my side.

kekiefer, 2021-02-15:
Since you really require singlewire for this to work, it's not really any better to check the board first, you can just try and fail later if it doesn't work when you try to set it up. This is what frsky does in `set_uart_single_wire`.

teyrana, 2021-02-20:
@kekiefer 

Okay, I'll give that a try.  I don't suppose there's another driver I could use as an example, is there? 

--- 

Still working on learning the code base -- particularly how board-specific hardware is defined.   Please bear with me :/ 

WIP Branch is at: 
https://github.com/teyrana/PX4-Autopilot/tree/pr-spektrum-telemetry

LorenzMeier, 2021-02-20:
The right example module is this one: https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/rc_input

You can see that two other telemetry systems have already been implemented. It also references this DSM parser implementation: https://github.com/PX4/PX4-Autopilot/tree/master/src/lib/rc

teyrana, 2021-02-21:
okay, so this is getting weird.   

the `rc_input` folder defines a module  (i.e. cmake -> `px4_add_module(...`)
AND the `frsky_telemetry` folder defines a module ( in the same way as `rc_input`) 
I'm assuming `spektrum telemetry` folder ought to define module as well? 

And if I understand the px4 architecture correctly,  the above modules [are essentially processes] (https://docs.px4.io/master/en/modules/module_template.html) right?  .... so each of the telemetry modules should behave independently ... right?  (the inherited patch changes code in the `rc_input` module, which doesn't track with the above understanding)

There's also an existing `spektrum_rc` module... architecturally, would it make sense to merge this patch with that module?  `src/drivers/spektrum_rc/*` vs `src/drivers/telemetry/spektrum_telemetry/*`   Does the px4 project try to separate the control (input) vs telemetry (output) communication paths?

teyrana, 2021-02-22:
@kekiefer 

I noticed there's multiple reference locations for the dsm parsers... is there a particular reason not to merge this patch with the existing `spektrum_rc` module?

kekiefer, 2021-02-22:
I'm not sure what you mean. IIRC the `spectrum_rc` task's job is to run the parser library that lives in `src/lib/rc/dsm.cpp`. This behavior wasn't changed, and the existing dsm parser was extended to support srxl for sending back telemetry.

teyrana, 2021-03-20:
@kekiefer 
Hey, how did you go about testing this?  
Should I add the `spektrum_telemetry` module to `boards/px4/fmu-v4/default.cmake`?   (or create a new file within that directory?)

All I really want to do is run that extra module, and debug the resulting messages. 

Also, it seems a bit much to upload the entire firmware.... is there a way I can just upload version of that module, and hot-swap it in?

teyrana, 2021-04-19:
## Update: 

I have a development branch .... building.   The debug port connects, and I can run 'rc_input', but the Spektrum receiver doesn't seem to be powered (I'm pretty sure it is from 'master'). 

I think the SPEKTRUM_POWER defines aren't being defined or set, properly.

teyrana, 2021-04-24:
@srodgers070 
Hi, thanks for the offer! 

The things that would be helpful are (1) understanding of the development board hardware at a detailed level , (2) willingness to code, and (3) flight testing ...

Unfortunately, I have not been able to reproduce either the telemetry or the control functionality, even on the bench, but when it's ready for testing, I will definitely post here.

teyrana, 2021-04-25:
Update: First bytes: 

I've gotten the fmu to read traffic from the receiver:
```
>>> Configure for SRXL...
    ::recv:: 14 SRXL bytes...
        ::Parsing: 14: [ A6 21 0E 10 70 0A 01 03 3B 4D 0B 1D 4D 5A]
        ::checksum:found:     4d5a:
        ::frame-type:  21:
```
If you're curious, the above is a handshake request frame from the receiver.

Unfortunately, the receiver won't do anything interesting until it receives a handshake reply... so I'll have to work on that before we see anything interesting.

teyrana, 2021-04-30:
@kekiefer 
1. Would you happen to remember which serial port you needed to use to write data back to the receiver? 
2. Also, which process were you using to test this?  (was it `rc_input` or `spektrum_telemetry` directly?)

kekiefer, 2021-04-30:
> Would you happen to remember which serial port you needed to use to write data back to the receiver?

I was using the DSM/SBUS RC port on the pixhawk 4. You'd have to look up what that is mapped to.

> Also, which process were you using to test this? (was it rc_input or spektrum_telemetry directly?)

I'm pretty sure it was all going through rc input, based on the fact that I had changes there.

teyrana, 2021-04-30:
Ah.  Okay, that's about where I got, as well :( 
I've verified reading from the DSM/SBUS RC port, but I don't seem to be able to write to it.  
(fwiw, rc_input seems to default to `/dev/ttyS3`)

I guess it's just too long ago, eh? :/

LorenzMeier, 2021-04-30:
Did you open the port for read/write? What is the error you are seeing? No error message but no messages on the logic analyzer?

teyrana, 2021-05-01:
## Update: It seems to be working

By "working", I mean: 
1. can receive messages, and recognize "handshake" frames
2. Can reply with a handshake-acknowledgement
3. Can then detect & monitor control frames
    - no heartbeat is detected; not sure if the receiver expects it or not.

## The settings I'm currently using are:
1. Open UART as line read/write (pre-existing code) 
    https://github.com/PX4/PX4-Autopilot/blob/4eb758edf0f0b8b5fd8a2bf06778930e1f9dec3c/src/lib/rc/dsm.cpp#L411
2. The single-wire configuration line. Copied from:
    https://github.com/PX4/PX4-Autopilot/blob/4eb758edf0f0b8b5fd8a2bf06778930e1f9dec3c/src/drivers/telemetry/frsky_telemetry/frsky_telemetry.cpp#L203

p.s. I don't have a logic analyzer.

----
Sample traffic: 
1. `[ A6 21 0E 10 71 0A 01 03 3B 4D 0B 1D 0A 89]`
    - This is a well-formed handshake request
2. `[ A6 21 0E 30 10 0A 00 00 12 34 56 78 0A 2C]`
    - this is a handshake acknowledgement
3. `[ A6 CD 0E 00 00 00 00 00 00 00 00 00 10 34]`
    - this is a well-formed control-channel frame
    - this frame is currently empty -- i.e. it is communicating zero channels.

teyrana, 2021-05-01:
@kekiefer 
Thanks :) 
I can confirm that running `rc_input` on the pixhawk4 mainboard works, and connects to the default port.

teyrana, 2021-05-02:
# For Discussion: 

1. `msg/input_rc.msg`->`channel_count`
This message lists `channel_count` as a count of all channels, but surrounding code _also_ treats it like a max-channel count... 
(eg. https://github.com/PX4/PX4-Autopilot/blob/165420598e820e87286004aecf808dfba40d3e3e/src/drivers/rc_input/RCInput.cpp#L185 )

    This assumption doesn't hold for the SRXL protocol, which may transmit non-continguous channels...  aka   channels 1,2, 10,11.

   - Is this a new design "constraint"?  Have you had discussions about this before? 
   - Do downstream consumers know to ignore "invalid" values in that message?

2. Channel level units -- 
The input_rc.msg file lists channel levels in units of PWM %  (in usec, specifically... I think)
https://github.com/PX4/PX4-Autopilot/blob/165420598e820e87286004aecf808dfba40d3e3e/msg/input_rc.msg#L36

Is the mapping between throttle percentage and pwm width explictily documented anywhere? 

I don't have a sufficient physical hardware to test this (i.e. airframes) and I wouldn't want to base any implementation off one airframe, anyways....

teyrana, 2021-05-03:
## Example parse log showing example parsing of control channels:

``` 
>>> Parse:SRXL: 26: [ A6 CD 1A 00 30 E5 00 00 89 0B 00 00 00 80 00 80 00 80 00 80 00 80 00 80 FB F6]
    >> Control Frame: 
             ::Reported-RSSI:     00e5  (   229)
             ::Channel-Mask:  00000b89  (  2953)
             ::Channels:  [ 0]: 8000 => 05dc
  [ 3]: 8000 => 05dc
channel invalid range: 0
  [ 7]: 0000 => 0000
channel invalid range: 1142
  [ 8]: 0476 => 0000
channel invalid range: 0
  [ 9]: 0000 => 0000
  [11]: 8000 => 05dc
        ....Parse successful...
>>> Parse:SRXL: 26: [ A6 CD 1A 00 00 64 00 00 76 04 00 00 A0 7F 00 80 40 D5 40 D5 00 80 00 80 89 C5]
INFO  [commander] Manual control regained after 0.9s
    >> Control Frame: 
             ::Reported-RSSI:     0064  (   100)
             ::Channel-Mask:  00000476  (  1142)
             ::Channels:  [ 1]: 8000 => 05dc
  [ 2]: d540 => 076a
  [ 4]: 8000 => 05dc
  [ 5]: 8000 => 05dc
  [ 6]: c589 => 0720
channel invalid range: 0
  [10]: 0000 => 0000
        ....Parse successful...
WARN  [rc_update] channel count changed 12 -> 11
```

kekiefer, 2021-05-03:
@teyrana I rebased https://github.com/kekiefer/Firmware/tree/pr-spektrum-telemetry in the last couple days, and now baseline reception of RC data via SPM4649T is working. Something is wrong with the way it plays with the telemetry task that I haven't looked into yet.

teyrana, 2021-05-04:
@kekiefer 
Okay, I was just trying to get RC Control to work first.  (I've barely even looked at telemetry, yet. )

Do you have any advice for how to scale the channels?

kekiefer, 2021-05-04:
The protocol for these receivers is handled with the same routines as for dsm -- `dsm_decode` takes care of decoding the received channel ID (up to 7 channels per frame, non-contiguous, in no particular order) to map them into the channels for rc input. Or maybe I don't understand what you're asking?

I'm curious if the code I sent will also work for your receiver. The telemetry task won't start by default, and so normal rc reception should work. In addition to fixing telemetry, the other thing that needs to be sorted is pairing the receiver. I left a routine `dsm_bind_srxl` that can be used, but there needs to be some way to call it.

teyrana, 2021-05-07:
> The protocol for these receivers is handled with the same routines as for dsm

I'm asking a more specific question -- what are the specifications, on the px4 side, for what the expected channel values are -- in a uorb "input_rc " msg?    The units are completely different -- what percentage of the pulse-width maps to what percentage of throttle (+/-)?   But the SRXL protocol seems to have different units than the "dsm" protocol?    (percentage throttle vs usec of pwm signal at ..... ¿50hz?)

Furthermore, [the existing code ]( https://github.com/PX4/PX4-Autopilot/blob/e265ebabc59575c1bd217f24b2d64a955aad1715/src/lib/rc/dsm.cpp#L147 )  is.... not the best documented.   It contains fields the SRXL spec doesn't -- [SRXL spec]( https://github.com/SpektrumRC/SRXL2/blob/master/Docs/SRXL2%20Specification.pdf ), p14: "7.7 Control Data Packet"  And also contains [magic numbers ](https://github.com/PX4/PX4-Autopilot/blob/e265ebabc59575c1bd217f24b2d64a955aad1715/src/lib/rc/dsm.cpp#L194) which don't apply to SRXL.... or at the very least, aren't apparent -- the limits in the existing dsm.cpp are applied after several transformations which are _also_ undocumented / uncited.

> the other thing that needs to be sorted is pairing the receiver. 
I don't get why that's a big deal?  there's a button right on the receiver to set that up? 

I mean, I have a pared-back branch that's working... I could just submit that, and let people start commenting on actual code... perhaps that would be more productive?

teyrana, 2021-06-19:
A PR is available that implements basic SRXL2 support: see #17555 .
Moving further discussion there.

mfry90, 2021-11-15:
Resolving this issue would be very helpful, since I'm not aware of Spektrum still selling SRXL (rather than SRXL2) receivers.

I hope this PR doesn't get abandoned, but would anyone else please post currently sold Spektrum receivers that can directly connect to the SBUS/DSM port without needing the SRXL2 feature?

teyrana, 2021-11-15:
@mfry90 
Would you believe that's what I was trying to do in the first place?  :/ 

I suggest you ask in the slack -- you'll get many more eyes on it, than here.