PX4/PX4-Autopilot #16627: Collision Prevention in SITL/Gazebo typhoon sim not function on 1.11.3

rencheckyoself, 2021-01-22:
**Describe the bug**
I am new to using PX4 so I have been working with the px4_sitl and gazebo simulation to test out some of the features. I wanted to test the collision prevention behavior so I added obstacles to the gazebo after the simulation launch, but the drone did not respond to the obstacles. I was using the typhoon_h480 simulation controlled using the virtual joystick in QGC. 

I think the bug is only present in 1.11.3 because I tested with 1.11.2 and 1.10.1 and the collision prevention prevented the drone from crashing in both of the previous versions.

I did not test on any drone hardware, only using the simulation and sitl.

**To Reproduce**
Steps to reproduce the behavior:
1. Installed ROS & Gazebo using the shell script provided in the PX4 documentation
2. Clone the current master branch of the PX4 software (or the 1.11.3 tag)
3. run `make px4_sitl gazebo_typhoon_h480`
4. Launch QCG and enable the virtual joystick
5. Insert a model with collision properties into the gazebo world.
8. Use Verify `CP_DIST` is positive (I used 2 meters) and Collision Prevention is enabled under the safety tab.
6. Use QGC to issue Take off
7. Switch Drone into position Mode
6. Use the joysticks to fly the drone toward the obstacle. 

**Expected behavior**
After seeing the behavior in previous versions, I expected the drone to stop based on the distance parameter.  

**Drone (please complete the following information):**
typhoon_h480 model

harshmittal2210, 2023-08-31:
Hi @rencheckyoself were you able to solve the collision prevention? 

I am trying in version 1.13.3 but it is not working properly.

rencheckyoself, 2023-08-31:
Hey, this was so long ago that I don’t remember. But yes in general the collision prevention should work, I was just having issues with this specific configuration.

closed 2023-08-31 by rencheckyoself