PX4/PX4-Autopilot #16628: [Question] RPi: All Wireless Communication via Ethernet?

Pskoog3, 2021-01-23:
Is it possible to put remote control, telemetry and video stream all on the Ethernet of raspberry pi with PX4 build?
I mean instead of connecting individual RC receiver and telemetry module to raspberry pi, all communication is routed via a Wifi modem connected to Ethernet port while the RC input is connected to QGroundControl or some other ground station. Is it already implemented?

Phillweston, 2021-01-27:
I think it is possible, but you need a companion computer running a specified embedded system, such as APSync or so.
Unfortunately, APSync does not support PX4 firmware at this time, and it is recommended to consider using Ardupilot firmware if you do have such a need.
You can refer to this instruction below:
https://ardupilot.org/dev/docs/apsync-intro.html

Phillweston, 2021-01-27:
You don't need to compile PX4-related source code on an on-board computer, which only transmits data to flight control through the mavlink protocol.

LorenzMeier, 2021-01-27:
Here is the documentation on how to set up the communication between the flight controller and the computer for forwarding via Wifi (or any other IP baed link): https://docs.px4.io/master/en/companion_computer/pixhawk_companion.html

closed 2021-01-27 by LorenzMeier