PX4/PX4-Autopilot #16630: print error messages(ERROR [navigator] vehicle_command lost) continuously use mavlink mount

liganggg, 2021-01-23:
**Describe the bug**
1.setting the param MNT_MODE_IN = MAVLINK_DO_MOUNT and MNT_MODE_OUT = MAVLINK
2.in the nsh can see it continuous printing of error messages, 
i found it from module navigator_main.cpp:
PX4_ERR("vehicle_command lost, generation %d -> %d", last_generation, _vehicle_command_sub.get_last_generation()); 
`ERROR [mavlink] vehicle_command lost, generation 0 -> 11                        
ERROR [navigator] vehicle_command lost, generation 5 -> 15                      
ERROR [navigator] vehicle_command lost, generation 15 -> 25                     
ERROR [navigator] vehicle_command lost, generation 25 -> 32                     
ERROR [navigator] vehicle_command lost, generation 32 -> 42                     
ERROR [navigator] vehicle_command lost, generation 42 -> 52                     
ERROR [navigator] vehicle_command lost, generation 52 -> 62                     
ERROR [navigator] vehicle_command lost, generation 62 -> 72                     
ERROR [navigator] vehicle_command lost, generation 72 -> 82                     
ERROR [navigator] vehicle_command lost, generation 82 -> 92  `

it means the orb 

> message (vehicle_command),publish too fast  in output_mavlink.cpp, 　and it lost some command at subscribe

I'm not sure if this is a BUG or there has  parameter setting error.
please give me some suggestions, 
thanks~

dayjaby, 2021-01-23:
I experienced the same and moved all gimbal commands into its own uORB topic. Have a look at https://github.com/dayjaby/Firmware/commit/9032de6bfb134161e19aa0387972dd0b07ab6390

I did not create a pull request as the whole gimbal control is being overhauled according to the mavlink gimbal protocol v2.

dagar, 2021-01-23:
There are relatively new checks in place anywhere `ORB_ID(vehicle_command)` is consumed to find dropped messages. In this particular case it's likely do to `vmount` publishing relatively fast.

Without more fundamental architectural changes around gimbal the fix right now would be a combination of...
1. making sure `vmount` doesn't publish `ORB_ID(vehicle_command)` faster than is necessary
2. increase `ORB_ID(vehicle_command)` queue depth https://github.com/PX4/PX4-Autopilot/blob/fa5a6cf712937a66cb07e3ba2682cf08347cad5b/msg/vehicle_command.msg#L137
3. add a loop in navigator so that it can process commands that being published faster than it's running https://github.com/PX4/PX4-Autopilot/blob/fa5a6cf712937a66cb07e3ba2682cf08347cad5b/src/modules/navigator/navigator_main.cpp#L215

closed 2021-01-26 by dagar