PX4/PX4-Autopilot #16631: Build Error px4_sitl_default

afiffadhlurrahman, 2021-01-23:
When i run `make px4_sitl_default gazebo`, this error appear, what should i do? Does the difference in ros affect the build? In my case, i use ros noetic and ubuntu 20.04.

```
-- PX4 version: v1.11.0-rc3-1232-g1d8808183d
-- PX4 config file: /home/afif/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
CMake Error at src/lib/CMakeLists.txt:48 (add_subdirectory):
  The source directory

    /home/afif/PX4-Autopilot/src/lib/ecl

  does not contain a CMakeLists.txt file.

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_airdata".
Call Stack (most recent call first):
  src/modules/airspeed_selector/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo_lookup".
Call Stack (most recent call first):
  src/modules/attitude_estimator_q/CMakeLists.txt:34 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/commander/CMakeLists.txt:37 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_EKF".
Call Stack (most recent call first):
  src/modules/ekf2/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/ekf2/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/local_position_estimator/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/mavlink/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/mc_pos_control/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/navigator/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/simulator/CMakeLists.txt:56 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/vmount/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo_lookup".
Call Stack (most recent call first):
  src/systemcmds/tests/CMakeLists.txt:88 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/examples/fake_magnetometer/CMakeLists.txt:34 (px4_add_module)

-- ROMFS: px4fmu_common
-- Configuring incomplete, errors occurred!
See also "/home/afif/PX4-Autopilot/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /home/afif/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [Makefile:219: px4_sitl_default] Error 1
```

Jaeyoung-Lim, 2021-01-23:
You didn't update the submodules
```
make submodulesupdate
```

afiffadhlurrahman, 2021-01-23:
I have run the `make submodulesupdate` command recently but its still get error when i try to build `make px4_sitl_default gazebo` @Jaeyoung-Lim 

```
Synchronizing submodule url for 'Tools/flightgear_bridge'
Synchronizing submodule url for 'Tools/jMAVSim'
Synchronizing submodule url for 'Tools/jsbsim_bridge'
Synchronizing submodule url for 'Tools/sitl_gazebo'
Synchronizing submodule url for 'boards/atlflight/cmake_hexagon'
Synchronizing submodule url for 'mavlink/include/mavlink/v2.0'
Synchronizing submodule url for 'platforms/nuttx/NuttX/apps'
Synchronizing submodule url for 'platforms/nuttx/NuttX/nuttx'
Synchronizing submodule url for 'platforms/qurt/dspal'
Synchronizing submodule url for 'src/drivers/gps/devices'
Synchronizing submodule url for 'src/drivers/uavcan/libuavcan'
Synchronizing submodule url for 'src/examples/gyro_fft/CMSIS_5'
Synchronizing submodule url for 'src/lib/ecl'
Synchronizing submodule url for 'src/lib/matrix'
Synchronizing submodule url for 'src/modules/micrortps_bridge/micro-CDR'
```

Jaeyoung-Lim, 2021-01-23:
@afiffadhlurrahman Okay, but the error should be different right?

afiffadhlurrahman, 2021-01-23:
The error after i run `make submodulesupdate`

```
-- PX4 version: v1.11.0-rc3-1232-g1d8808183d
-- PX4 config file: /home/afif/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
CMake Error at src/lib/CMakeLists.txt:48 (add_subdirectory):
  The source directory

    /home/afif/PX4-Autopilot/src/lib/ecl

  does not contain a CMakeLists.txt file.

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_airdata".
Call Stack (most recent call first):
  src/modules/airspeed_selector/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo_lookup".
Call Stack (most recent call first):
  src/modules/attitude_estimator_q/CMakeLists.txt:34 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/commander/CMakeLists.txt:37 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_EKF".
Call Stack (most recent call first):
  src/modules/ekf2/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/ekf2/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/local_position_estimator/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/mavlink/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/mc_pos_control/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/navigator/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/simulator/CMakeLists.txt:56 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/modules/vmount/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo_lookup".
Call Stack (most recent call first):
  src/systemcmds/tests/CMakeLists.txt:88 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
  get_target_property() called with non-existent target "ecl_geo".
Call Stack (most recent call first):
  src/examples/fake_magnetometer/CMakeLists.txt:34 (px4_add_module)

-- ROMFS: px4fmu_common
-- Configuring incomplete, errors occurred!
See also "/home/afif/PX4-Autopilot/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /home/afif/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [Makefile:219: px4_sitl_default] Error 1
```

Jaeyoung-Lim, 2021-01-23:
@afiffadhlurrahman Are you sure the submodule update have been done correctly? Have you checked if the Cmakefle.txt actually exists in the location that it is complaining?

afiffadhlurrahman, 2021-01-24:
Okay, i will try it later, thanks

afiffadhlurrahman, 2021-01-28:
Thanks, my simulator now works, i re-clone px4 repo and run submodule again @Jaeyoung-Lim

closed 2021-01-28 by afiffadhlurrahman