PX4/PX4-Autopilot #16639: CI SITL plane failures

dagar, 2021-01-24:
SITL plane has been failing lately in CI. https://github.com/PX4/PX4-Autopilot/pull/16635/checks?check_run_id=1758446359

[image]
[image]

Jaeyoung-Lim, 2021-01-25:
Looks like there were some changes in the liftdrag plugin: https://github.com/PX4/PX4-SITL_gazebo/pull/562

sfuhrer, 2021-01-25:
Look to me like an issue in the FW takeoff - manual flying, GOTOs and landings look fine, but 100% reproducible when you start a mission with a takeoff. Probably something pitch setpoint related. 
[image]

mrpollo, 2021-04-28:
fixed by #16792 

Thanks @sfuhrer

closed 2021-04-28 by mrpollo