PX4/PX4-Autopilot #16643: review wind_estimate message architecture

dagar, 2021-01-25:
- each ekf2 instance publishes an instance of `wind_estimate`
 - the airspeed selector publishes N+1 `wind_estimate` messages
 - RTL and mavlink naively consume the first instance of `wind_estimate`, but that might not be correct

dagar, 2021-01-25:
FYI @jkflying

dagar, 2021-01-25:
Proposal that initially comes to mind assuming that we'll prefer the wind_estimate from ekf2 for these end users. Correct me if I'm wrong, but I don't think we'd have a real scenario where things like RTL would prefer using the `wind_estimate` from a standalone wind estimator?
1. airspeed selector log to different messages entirely 
2. With multi-ekf the ekf2selector becomes responsible for publishing the one and only wind_estimate. It either recreates it directly from data in estimator_states, etc or we replicate the pattern for other messages. 
    - eg N x estimator_attitude => vehicle_attitude

sfuhrer, 2021-01-28:
>Correct me if I'm wrong, but I don't think we'd have a real scenario where things like RTL would prefer using the wind_estimate from a standalone wind estimator?
I think it just doesn't matter that much where it comes from. Recent flight testing in some wind have shown that the wind estimates from EKF and the airspeed_selector are almost identical (not surprising with it having the same setting, but still nice to see).

[image]

I think using the one from EKF as you propose is feasible for the current consumers of wind_estimate (high wind detection, displaying it on the grounstation, using in the range-based RTL) - but it's probably too low-passed filtered for applications that also need to see gusts - like the windy L1 controller of @tstastny that we currently again are working on to bring in. 
I would propose to use a new wind estimator for it ("gusty wind estimator" or what ever we want to call it), with separate tuning values (at the same time, I think the wind estimators inside airspeed_selector could just use the ones from EKF, unless you think using EKF2_* params outside of EKF isn't an option)). In my eyes, it could either live directly in the FW_position_controller module, or we refit the `airspeed_selector` into an `airdata` module that also publishes this wind estimate or maybe @dagar has a more elegant solution for it's placement.

sfuhrer, 2021-02-04:
@dagar If you confirm that:

> 1. airspeed selector log to different messages entirely
>
>     2. With multi-ekf the ekf2selector becomes responsible for publishing the one and only wind_estimate. It either recreates it directly [...]

is still your preferred way forward (it would be mine) then I can go ahead and make the changes required.

dagar, 2021-02-04:
> is still your preferred way forward (it would be mine) then I can go ahead and make the changes required.

Yes, let's do it.

closed 2021-02-20 by dagar