PX4/PX4-Autopilot #16659: actuator_controls_0 is not a normalized value

dallinbriggs, 2021-01-26:
**Describe the bug**
According to[ the documentation](https://docs.px4.io/master/en/concept/mixing.html) of control group #0, all of the flight controls should be normalized values between -1 and 1. During a flight test that ended with instability and then a crash of a hexacopter, the values in `actuator_controls_0` can be seen exceeding the value of 1. Since consumers of `actuator_controls_0` may be assuming a normalized value, I think this could lead to other unanticipated errors.

**To Reproduce**
I haven't been able to reproduce the error outside of this flight.

**Expected behavior**
The values of `actuator_controls_0` to saturate at 1 or -1.

**Log Files and Screenshots**
[Flight Report](https://review.px4.io/plot_app?log=a65a5721-5bac-4d04-8642-576cc7fe7125)
[Log file](https://drive.google.com/file/d/1YX4sl5-zYj6PMtfHPPXkbYwCmfrKJCPd/view?usp=sharing)
[Discussion](https://discuss.px4.io/t/possible-software-bug-causing-crash/20523) about the crash.

**Drone:**
Airframe - Tarot T960
Pixhawk - Holybro Pixhawk 4 mini
Motors - Sunnysky X4112S 400 Kv
ESC - APD F series 80 amp
Props - T-motor carbon 16x5.4"
Battery - 6s 16000 mah Tattu smart battery
[image]

**Additional context**
I don't think this bug caused the crash, but it may be a symptom of something further up the chain. I'm still working to determine the cause of the crash. Below is the `actuator_controls_0` plot from the flight. This image is the last few seconds of the flight when the aircraft went unstable.
[image]

bresch, 2021-01-27:
Looks like a failure of the rear right motor to me: the motor 3 command is at full thrust, the pitch and roll controller are commanding a negative value, the commanded thrust is at max and the drone is still going down.

Regarding the [-1; 1] saturation of the actuator controls, it is actually performed in the mixer, after the `actuator_controls` message is published and logged. In your case, the large actuator control values are a consequence of a failure and not the cause of it.

dallinbriggs, 2021-01-27:
@bresch I also thought it might be a motor failure, but then I rebuilt the drone with all new ESC's, motors, and a different flight controller (the cube orange), and the drone almost crashed again due to instability in position hold mode. It seems unlikely that two different motor and ESC sets would have a failure.

[Here is the link](https://review.px4.io/plot_app?log=3e1adf31-1eba-4c3b-ac79-d52b223548b3) to the other flight that was stopped after a weird instability was noticed after switching into position hold mode. Although the actuator controls did not exceed the value of 1, a similar behavior happened where the setpoints and the estimated response were almost going in the opposite directions. Again, this second flight was with brand new motors and ESC's with a cube orange flight controller.

dallinbriggs, 2021-02-18:
After a lot of trial and error, I found that I was having intermittent motor desync issues with random motors. This looks to be the cause of the crash. I still need to investigate though, if the values in actuator_controls_0 really can be non-normalized.

julianoes, 2023-04-03:
Based on the fact there was no more follow up I think this can be closed.

closed 2023-04-03 by julianoes