PX4/PX4-Autopilot #16670: HITL Simulation very jumpy

AbhishekGS, 2021-01-27:
**Describe the bug**
I performed HITL Simulation with Pixhawk 1 hardware using 1.10.2 firmware version with jMAVsim simulator. The controller performance for a simple mission was very jumpy and not smooth at all.  

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the [1.10.2 firmware](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.10.2) on Pixhawk 1.
2. Follow the steps mentioned in this [doc](https://docs.px4.io/master/en/simulation/hitl.html#setting-up-hitl)  on how to set up the QGC for HITL.
3. Follow the steps mentioned [here](https://docs.px4.io/master/en/simulation/hitl.html#jmavsim-quadrotor-only) for starting the jMAVSim simulator in HITL mode.
4. Create a simple mission, upload it and start. Follow the instructions mentioned [here](https://docs.px4.io/master/en/simulation/hitl.html#fly-an-autonomous-mission-in-hitl) .
5. Observe the jumpy behaviour

**Expected behavior**
I expect a smooth controller performance. 

**Log Files and Screenshots**
- The log flight review can be found [here](https://review.px4.io/plot_app?log=ef92140a-b1bc-4c82-80c6-48ccb6224838). 
- The video of the quadcopter performing the mission in jMAVSim can be seen [here](https://drive.google.com/file/d/13PI-148FiCjFmpaoCSy16xiPDDERk-RD/view?usp=sharing).

julianoes, 2021-04-29:
This might be related to #17006.

julianoes, 2021-05-02:
Closing as duplicate.

closed 2021-05-02 by julianoes