PX4/PX4-Autopilot #16677: parameter 'expected' set but not used [-Werror=unused-but-set-parameter]

yangshaojinyangshaojin, 2021-01-28:
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[846/1379] Building CXX object src/drivers/imu/invensense/icm20689/CMakeFiles/drivers__imu__invensense__icm20689.dir/ICM20689.cpp.obj
FAILED: src/drivers/imu/invensense/icm20689/CMakeFiles/drivers__imu__invensense__icm20689.dir/ICM20689.cpp.obj 
/usr/bin/ccache /usr/bin/arm-none-eabi-g++  -DCONFIG_ARCH_BOARD_PX4_FMU_V5 -DMODULE_NAME=\"icm20689\" -DPX4_MAIN=icm20689_app_main -D_SYS_CDEFS_H_ -D_SYS_REENT_H_ -D__CUSTOM_FILE_IO__ -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem ../../platforms/nuttx/NuttX/include/cxx -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -I../../boards/px4/fmu-v5/src -I../../platforms/nuttx/src/px4/common/include -I. -Isrc/lib -I../../platforms/nuttx/src/px4/stm/stm32f7/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -mcpu=cortex-m7 -mthumb -mfpu=fpv5-d16 -mfloat-abi=hard -Os -DNDEBUG   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -nostdinc++ -fno-sized-deallocation -Wframe-larger-than=2048 -std=gnu++14 -MD -MT src/drivers/imu/invensense/icm20689/CMakeFiles/drivers__imu__invensense__icm20689.dir/ICM20689.cpp.obj -MF src/drivers/imu/invensense/icm20689/CMakeFiles/drivers__imu__invensense__icm20689.dir/ICM20689.cpp.obj.d -o src/drivers/imu/invensense/icm20689/CMakeFiles/drivers__imu__invensense__icm20689.dir/ICM20689.cpp.obj -c ../../src/drivers/imu/invensense/icm20689/ICM20689.cpp
In file included from ../../platforms/common/include/px4_platform_common/atomic_bitset.h:36:0,
                 from ../../src/modules/uORB/uORBDeviceMaster.hpp:54,
                 from ../../src/modules/uORB/uORBDeviceNode.hpp:37,
                 from ../../src/modules/uORB/Publication.hpp:45,
                 from ../../src/modules/uORB/PublicationMulti.hpp:45,
                 from ../../src/lib/drivers/accelerometer/PX4Accelerometer.hpp:40,
                 from ../../src/drivers/imu/invensense/icm20689/ICM20689.hpp:46,
                 from ../../src/drivers/imu/invensense/icm20689/ICM20689.cpp:34:
../../platforms/common/include/px4_platform_common/atomic.h: In instantiation of 'bool px4::atomic<T>::compare_exchange(T*, T) [with T = unsigned int]':
../../src/drivers/imu/invensense/icm20689/ICM20689.cpp:382:79:   required from here
../../platforms/common/include/px4_platform_common/atomic.h:169:34: error: parameter 'expected' set but not used [-Werror=unused-but-set-parameter]
  inline bool compare_exchange(T *expected, T desired)
                                  ^~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[848/1379] Building CXX object src/dri...drivers__roboclaw.dir/RoboClaw.cpp.obj
ninja: build stopped: subcommand failed.
Makefile:217: recipe for target 'px4_fmu-v5_default' failed
make: *** [px4_fmu-v5_default] Error 1

Thanks!

dagar, 2021-02-01:
Can you check your toolchain (arm-none-eabi-gcc) version? http://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html

dagar, 2021-02-02:
I believe this will be resolved if you use the same toolchain versions as the documentation. Please re-open if that's not correct.

closed 2021-02-02 by dagar