PX4/PX4-Autopilot #16679: Geofence issues

garfieldG, 2021-01-28:
**Describe the bug**
Hi,

While playing with the geofence on sitl I encountered  few issues regarding it :
1. Altitude - while on geofence failsafe trigger hold mode I tried to change the altitude to be above the allowed geofence max altitude. The first time trying to do so it warned me about the altitude geofence and didn't go as high as I wanted it to, but then when I tried to give a new altitude command to go above the allowed max altitude it did reach my desired altitude, although it shouldn't.
2.  I tried to breach the geofence and in geofence, none mode fly to a waypoint after the max geofence radius, later I switched back to geofence hold mode, and the mc flow back to a waypoint inside the allowed radius, as it should.
The problem was when I did it all the same when the altitude of the mc was above the allowed max altitude, the stayed in it's waypoint outside the allowed radius and didn't fly back to the allowed radius.
3. Although the geofence mode was none, trying to upload a mission with a waypoint above the max radius failed due to geofence violation in one of the mission waypoints.

**To Reproduce**
Steps to reproduce the behavior:
first issue :
1. Change geofence mode to hold, the max altitude to 50m (can be something else) and the max radius to 30m (can be something else).
2. Change altitude to 60m. It should stop ascending a little above 50m and warn about geofence violation.
3. Change altitude to 60m. It should get to 60m while warning about geofence violation
second issue :
4. Change geofence mode to none.
5. Fly to a waypoint outside the max radius.
6. Switch the geofence mode to hold.
7. see the error.

third issue :
1. change geofence mode to none.
2. try to upload a mission with one of its waypoints outside the max radius.
3. mission check will fail due to geofence violation in one of the waypoints.

**Expected behavior**
first issue : mc shouldn't go above the allowed altitude.
second issue : mc should return to the allowed radius.
third issue : mission should be accepted.

**Drone (please complete the following information):**
- (px4 master)[ https://github.com/PX4/PX4-Autopilot/tree/e014954c91a14656f6922756f6cfc020d2da824f].
- gzebo (iris model, empty word).
- qgc version v4.1.1.

garfieldG, 2021-07-06:
anything new?

RomanBapst, 2021-07-06:
@garfieldG Hi, the fix for the vertical geofence is this one here:
https://github.com/PX4/PX4-Autopilot/pull/17566

Some checks did not pass so I rebased on master and pushed gain, should be merged soon.
Can you please retest with that one?

RomanBapst, 2021-07-06:
@garfieldG 

> Change geofence mode to none.
Fly to a waypoint outside the max radius.
Switch the geofence mode to hold.
see the error.

I don't think this is an error, it's just not supported this way. If you have geofence deactivated and you fly outside of the geofence you cannot expect the system to bring you back into the geofence. In some cases this might actually be dangerous and not expected. If you want to use a geofence, just have it enabled always.

garfieldG, 2021-08-02:
I think it's an error as long you can change the gf action, otherwise the action should be set once and after that you shouldn't be able to change it