PX4/PX4-Autopilot #16681: JSBSim Shared Pointer issue with bridge

xxEoD2242, 2021-01-28:
When trying to build jsbsim_rascal with px4_sitl, I got the following error:

```
PX4-Autopilot/Tools/jsbsim_bridge/src/jsbsim_bridge.cpp:146:33: error: no viable conversion from 'std::shared_ptr<FGInitialCondition>' to 'JSBSim::FGInitialCondition *'
    JSBSim::FGInitialCondition *initial_condition = _fdmexec->GetIC();
                                ^                   ~~~~~~~~~~~~~~~~~
```

As the compiler says, this is a valid error, since this is a standard pointer and not a shared pointer. To solve this, I did the following:

```
    std::shared_ptr<JSBSim::FGInitialCondition> initial_condition = _fdmexec->GetIC()
```

Once I did this, everything compiled and the system worked just fine. Now, this was a naive attempt to try to fix what was wrong and JSBSim seems to be working just fine with QGC.

Platform: MacOSx Big Sur 11.1
PX4 Version: Latest Pull
JBSim Version: 1.2.2

I'd probably say this is way beyond my level to do a pull request on but hope this helps someone.

**To Reproduce:**
Run `make px4_sitl jsbsim_rascal` on a MacOS with Big Sur

**Expected behavior**
JSBSim and PX4 to launch properly.

**Drone (please complete the following information):**
- PX4 Sitl

Jaeyoung-Lim, 2021-01-28:
@xxEoD2242 This should be fixed in https://github.com/PX4/PX4-Autopilot/pull/16685

Would you be able to give it a try? Thanks!

closed 2021-01-29 by bkueng

xxEoD2242, 2021-01-29:
@Jaeyoung-Lim This worked! Thanks!