PX4/PX4-Autopilot #16686: PX4Flow does not start with the specified I2C bus type using Firmware 1.11.

acarrou, 2021-01-28:
PX4Flow does not start with the specified I2C bus type using Firmware 1.11.

I have tried the PX4Flow on the PX4 1.11 firmware and there is an issue with the address being assigned. I tried this with firmware version 1.10 and it seems to have its default address of 0x42.  The error I am getting is:
```ERROR [SPI_I2C] need to specify a bus type```

I have the driver needed as well as all the necessary parameters enabled for this.

**To Reproduce**
Steps to reproduce the behavior:
1. Using firmware 1.11, start the PX4Flow by going into the MavLink Console and type '''px4flow start''' the error will show that there is not a specified bus.

**Expected behavior**
When the Pixhawk turns on the PX4Flow should be feeding data to the Pixhawk when it turns on, it should not need to have the address set, like in firmware 1.10.

```
px4flow start
ERROR [SPI_I2C] need to specify a bus type
Usage: px4flow <command> [arguments...]
 Commands:

   start
     [-I]        Internal I2C bus(es)
     [-X]        External I2C bus(es)
     [-b <val>]  bus (board-specific internal (default=all) or n-th external (default=1))
     [-f <val>]  bus frequency in kHz
     [-a <val>]  I2C address
                 default: 66
     [-R <val>]  Rotation (default=downwards)
                 default: 25

   stop

   status        print status info
nsh>
```

**Additional context**
I am able to get the address to work if I put in the command
```px4flow start -X -a 0x42```
to set the address but I have to do this every time I turn the pixhawk on.

bresch, 2021-02-04:
Did you set `SENS_EN_PX4FLOW` ?
Ref: https://github.com/PX4/PX4-Autopilot/issues/14043#issuecomment-579300002

acarrou, 2021-02-04:
@bresch Yes, I did. I actually rolled back to 1.10.1 and it works on boot.