PX4/PX4-Autopilot #16692: MRO Control Zero i2c issue

ryanjAA, 2021-01-29:
When testing the Control Zero F7 there are some problem with the i2c bus. Namely that data seems to "jump" i.e. not be stable.

An HMC5883 mag, MS4525 airspeed sensor and Lightware sf11/c Lidar rangefinder were all connected via i2c. Both with internal power and externally powered. Subsequent tests were done with just the lidar directly connected to the autopilot to remove any potential interference issues from the made harnesses (but data output was the same, no induced interference but the aim was to limit variables).

The lidar data hops around +/- 0.30m on the bus. Airspeed is hard to tell since on the bench the sensor normally moves up and down and the mag data movement is TBD but the lidar data jumping around is also seen when flight tested (and likely other data) although with the slower logging rate due to EKF2 replay now disabled by default (thanks @dagar for pointing that out), it is difficult to tell from the captured logs if the other data is jumping as much.

Nonetheless this poses a large issue given there isn't a reason for this to occur from what I can tell on the software. Master, release 1.11.3 as well as Ardupilot were tested. Same issue. If you switch to a different Pixhawk (1 or 4), it goes away and is rock solid, i.e. it doesn't move more than 0.01m. I tested 6 different lidars. All on the same lidar firmware and same type and all fine on Pixhawk 1 and Pixhawk 4 just not the Control Zero. 

It's important to note this since using external sensors (on i2c) is crucial for most and if this can't be relied on then there are serious implications for flight.

I put a logic analyzer on the i2c bus with just the lidar connected on both a Pixhawk 1 and the Control Zero. The data is [here](https://files.appliedaeronautics.com/_pQ5JFblQMVBMBR) in Saleae logic format. 

There was quite a bit of throttle surging (also viewable in the below log) and while not sure, I do questions if this was due to airspeed data jumps (this foam aircraft is not used too often although on the previous use with a different autopilot [Pixhawk 1] on the exact plane and hardware this did not occur). 

Anyway - I'd say until this is remedied or ascertained to be ok, it is not ok. 

The gif at the bottom is of the data jumping and attached is a flight log on release 1.11.3 with the RSSI and PPM fix pulled in.

[image]

Also, this doesn't belong in https://github.com/PX4/PX4-Autopilot/issues/16569 as that is for support overall and this seems to be a potential hardware bug.

https://review.px4.io/plot_app?log=968de235-6605-4353-904c-d2bc105cb018

mrpollo, 2021-02-02:
Thanks for reporting @ryanjAA

ryanjAA, 2021-02-26:
Closing. Replacement autopilot hardware and issue is gone. 

I was actually able to eventually get the existing hardware to stop doing it by letting the autopilot run almost the entire day (they run hot so all day at operating temp) and then let cool overnight and it doesn’t do it anymore but it definitely had to  “burn in”. 

The comm errors issue is still there but I think that is likely a PX4 issue and not hardware related the more this is tested.

closed 2021-02-26 by ryanjAA