PX4/PX4-Autopilot #16725: LSM9DS1 Accel/Gyro and Mag on RPi SPI

Pskoog3, 2021-02-01:
I am trying to interface LSM9DS1 accel/gyro and mag on raspberry pi 3 SPI with accel/gyro CS connected to CE0 of RPi and mag on CE1. It seems only CE0 is responsive. 

When accel/gyro CS is connected to CE0 and mag CS to CE1, only lsm9ds1 status is ok and lsm9ds1_mag status fails. Alternatively when mag CS is connected to CE0, lsm9ds1_mag status is fine but lsm9ds1 status fails for accel/gyro CS connected to CE1.

I am using the following configuration.
lsm9ds1 -s -R 0 start
lsm9ds1_mag -s -R 0 start

dagar, 2021-02-01:
How is the SPI bus configured? 

https://github.com/PX4/PX4-Autopilot/blob/0c74028976ef13697522bacbc6ac13effc898db3/boards/emlid/navio2/src/spi.cpp#L37-L44

Pskoog3, 2021-02-02:
> How is the SPI bus configured?
> 
> https://github.com/PX4/PX4-Autopilot/blob/0c74028976ef13697522bacbc6ac13effc898db3/boards/emlid/navio2/src/spi.cpp#L37-L44

I don't understand what the second argument number means in initSPIDevice? Can you please elaborate.

dagar, 2021-02-02:
> Does the 1st argument in initSPIBus is the SPI0 (/dev/spidev0.x) and the 2nd argument in initSPIDevice is the chip enable (CE), (/dev/spidev0.2 for LSM9DS1_M)??

It's the Linux spidev numbering. PX4 SPI on Linux doesn't touch the chip selects directly, it simply interacts with /dev/spidevX.Y.

The first group (`initSPIBus(0,`) is the bus (/dev/spidev0), then each `initSPIDevice` entry is the chip select (/dev/spidev0.X). What that corresponds to physically usually comes from a device tree overlay.

Pskoog3, 2021-02-02:
Got it working... thanks. However, LSM9DS1 seems to be too sensitive and couldn't get the sensors calibrated! Might be the IMU itself is defective.

closed 2021-02-02 by Pskoog3

dagar, 2021-02-02:
> Got it working... thanks. However, LSM9DS1 seems to be too sensitive and couldn't get the sensors calibrated! Might be the IMU itself is defective.

If you can capture a flight log from boot with the attempted calibration I'll take a look. The data would have to be extremely bad for calibration to not be possible.

Pskoog3, 2021-02-03:
I was using MPU6500 before, and because of sudden change to LSM9DS1 perhaps, the system is unable to calibrate with new range and sensitivity... is that possible? If so, how can rest the onboard system as well as QGroundControl?

dagar, 2021-02-03:
I could see it being a problem for the level calibration, but not for accel & gyro calibration. It might be obvious from a log. https://logs.px4.io/

Pskoog3, 2021-02-03:
Yes you are correct... accel and gyro calibration completes successfully, but too sensitive and level calibration is indeed a problem.
How can I do a clean accel, gyro and level calibration and reset the system?

Pskoog3, 2021-02-06:
> I could see it being a problem for the level calibration, but not for accel & gyro calibration. It might be obvious from a log. https://logs.px4.io/

The log is available at the following link... was unable to upload to logs.px4.io
https://drive.google.com/file/d/1UAWU1LU9kpD4hHc-K8WDeF2FWpS83VWN/view?usp=sharing
Either the sensor itself is faulty or driver needs to be updated!