PX4/PX4-Autopilot #16735: Dshot telemetry dropouts

benjinne, 2021-02-02:
**Describe the bug**
On my Quadcopter running the latest PX4, each ESC telemetry reading periodically report zero when using the APD 120F3 ESCs.
The issue does not occur on my other quadcopter using Holybro Tekko32 ESCs. This is because the APD ESCs undergo a desync routine that periodically prevents the telemetry from updating (As what was described to me since I've been in contact with APD).

**To Reproduce**
Steps to reproduce the behavior:
1. Enable DShot telem then either watch the esc status uorb in QGC or capture a ulog
2. Observe ESC voltage and current readings periodically record zero

**Expected behavior**
Voltage and Current readings should never report zero when powered on

**Log Files and Screenshots**
Each down spike on both plots is a packet containing 10 bytes which I can observe when I zoom in. The zoomed out picture shows the inconsistency in the APD ESCs vs the Tekko32 ESCs. 

APD 120F3 telem 500ms capture
[image]
Tekko32 telem 20ms capture
[image]

bkueng, 2021-02-03:
So what do you expect the autopilot to do instead? There is a checksum and when there's no response, it will mark the ESC as offline (and reset the voltage).

benjinne, 2021-02-03:
APD told me the last valid value should be repeated

benjinne, 2021-02-03:
Direct quote "the ESC's will forgo the telemetry response for a few cycles depending on the timing, and it prioritises motor control. The other FC stacks get around this by persisting the last value of the telemetry"

bkueng, 2021-02-04:
From the image it looks like you don't get any response at all (or do you just get packets with 0 values for voltage + current?). If so, how do you know the ESC didn't go offline?
Repeating stale data is just misleading and could cover real issues.

benjinne, 2021-02-04:
I am getting no response what seems randomly, and it seems to make sense according to APD. I agree repeating stale data is problematic. I sent APD this issue and we'll see their response.

benjinne, 2021-05-18:
Update: APD claims they are working on releasing a version that fixes this issue. I have yet to try it out but hopefully it will work

galdalali, 2021-08-11:
Hey, have you managed to solve this issue?

benjinne, 2021-08-11:
> Hey, have you managed to solve this issue?

I don't know, but APD told me they would update their firmware at some point. I haven't heard from them since I last requested a working version with PX4.

benjinne, 2021-08-11:
The latest firmware for the F series  is 2.0.0, found here https://docs.powerdrives.net/downloads/firmware/firmware-releases Let me know if this works for you

Vanja-APD, 2022-04-06:
Improved telemetry response was one of the focuses of our Rev 2.1.0 Beta firmware, details here:

https://docs.powerdrives.net/downloads/firmware/firmware-releases/f-series-beta-firmware#2.0.0-19th-february-2021-latest-2

Once Beta testing is complete, this will be the standard released functionality.

bresch, 2023-03-03:
I tried recently with a APD 80F3x (newest firmware) and PX4 main: no issues.
[image]