PX4/PX4-Autopilot #16745: MAVSDK Tailsitter tests timeout

mrpollo, 2021-02-03:
I started seeing a few PRs getting stuck on CI, specifically on the MAVSDK tests for Tailsitter, its very common to see the error below

Happens on this test:
```
test/mavsdk_tests/mavsdk_test_runner.py --speed-factor 20 --abort-early --model tailsitter test/mavsdk_tests/configs/sitl.json
```

Output;
```
[mavsdk_tests] [03:40:39|Debug] MAVLink: info: [logger] ./log/2021-02-02/15_40_40.ulg (system_impl.cpp:229)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Land on GPS lost during mission (baro height mode)
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] ../../../test/mavsdk_tests/test_multicopter_failsafe.cpp:37
[mavsdk_tests] ...............................................................................
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   REQUIRE( Mission::Result::Success == result )
[mavsdk_tests] with expansion:
[mavsdk_tests]   Success == Timeout
[mavsdk_tests] terminate called after throwing an instance of 'Catch::TestFailureException'
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   {Unknown expression after the reported line}
[mavsdk_tests] due to a fatal error condition:
[mavsdk_tests]   SIGABRT - Abort (abnormal termination) signal
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases: 1 | 1 failed
[mavsdk_tests] assertions: 8 | 6 passed | 2 failed
[px4         ] INFO  [tone_alarm] notify neutral
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
[gzserver    ] 
```

mrpollo, 2021-02-03:
Here is one more fail on CI

https://github.com/PX4/PX4-Autopilot/pull/16723/checks?check_run_id=1815729601

```
[mavsdk_tests] [03:40:39|Debug] MAVLink: info: [logger] ./log/2021-02-02/15_40_40.ulg (system_impl.cpp:229)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Land on GPS lost during mission (baro height mode)
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] ../../../test/mavsdk_tests/test_multicopter_failsafe.cpp:37
[mavsdk_tests] ...............................................................................
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   REQUIRE( Mission::Result::Success == result )
[mavsdk_tests] with expansion:
[mavsdk_tests]   Success == Timeout
[mavsdk_tests] terminate called after throwing an instance of 'Catch::TestFailureException'
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   {Unknown expression after the reported line}
[mavsdk_tests] due to a fatal error condition:
[mavsdk_tests]   SIGABRT - Abort (abnormal termination) signal
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases: 1 | 1 failed
[mavsdk_tests] assertions: 8 | 6 passed | 2 failed
[px4         ] INFO  [tone_alarm] notify neutral
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
[gzserver    ] 
```

mrpollo, 2021-02-15:
One more timeout CI failure

https://github.com/PX4/PX4-Autopilot/pull/16695/checks?check_run_id=1827591700

```
[mavsdk_tests] [02:27:08|Info ] New system on: 127.0.0.1:14580 (with sysid: 1) (udp_connection.cpp:190)
[mavsdk_tests] [02:27:08|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:416)
[mavsdk_tests] [02:27:08|Debug] Component Autopilot (1) added. (system_impl.cpp:315)
[mavsdk_tests] [02:27:09|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:488)
[mavsdk_tests] [02:27:09|Debug] MAVLink: info: [logger] ./log/2021-02-04/02_27_11.ulg (system_impl.cpp:229)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Land on GPS lost during mission (baro height mode)
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] ../../../test/mavsdk_tests/test_multicopter_failsafe.cpp:37
[mavsdk_tests] ...............................................................................
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   REQUIRE( Mission::Result::Success == result )
[mavsdk_tests] with expansion:
[mavsdk_tests]   Success == Timeout
[mavsdk_tests] terminate called after throwing an instance of 'Catch::TestFailureException'
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   {Unknown expression after the reported line}
[mavsdk_tests] due to a fatal error condition:
[mavsdk_tests]   SIGABRT - Abort (abnormal termination) signal
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases: 1 | 1 failed
[mavsdk_tests] assertions: 8 | 6 passed | 2 failed
[px4         ] WARN  [mavlink] Operation timeout
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
[gzserver    ] [Msg] Using MAVLink protocol v2.0
[gzserver    ] recvfrom error: Connection reset by peer
[gzserver    ] Connection closed by client.
```

mrpollo, 2021-02-15:
Here's another CI timeout on a MAVSDK test, this one is on multicopter

https://github.com/PX4/PX4-Autopilot/pull/16255/checks?check_run_id=1438461543

```
[mavsdk_tests] [02:32:49|Info ] New system on: 127.0.0.1:14580 (with sysid: 1) (udp_connection.cpp:190)
[mavsdk_tests] [02:32:49|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:416)
[mavsdk_tests] [02:32:50|Debug] Component Autopilot (1) added. (system_impl.cpp:315)
[mavsdk_tests] [02:32:50|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:488)
[mavsdk_tests] [02:32:50|Debug] MAVLink: info: [logger] ./log/2020-11-22/14_32_51.ulg (system_impl.cpp:229)
[mavsdk_tests] Waiting for system to be ready
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Land on GPS lost during mission (baro height mode)
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] ../../../test/mavsdk_tests/test_multicopter_failsafe.cpp:37
[mavsdk_tests] ...............................................................................
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   REQUIRE( Mission::Result::Success == result )
[mavsdk_tests] with expansion:
[mavsdk_tests]   Success == Timeout
[mavsdk_tests] terminate called after throwing an instance of 'Catch::TestFailureException'
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:187: FAILED:
[mavsdk_tests]   {Unknown expression after the reported line}
[mavsdk_tests] due to a fatal error condition:
[mavsdk_tests]   SIGABRT - Abort (abnormal termination) signal
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases: 1 | 1 failed
[mavsdk_tests] assertions: 8 | 6 passed | 2 failed
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
```

mrpollo, 2021-02-15:
@julianoes I thought you might be interested in these, I don't know how to proceed here, my only guess is that the GH Containers have some sort of constraining that we seem to be randomly hitting.

mrpollo, 2021-02-15:
One more timeout

https://github.com/PX4/PX4-Autopilot/runs/1905699580?check_suite_focus=true
```
[mavsdk_tests] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[mavsdk_tests] mavsdk_tests is a Catch v2.9.2 host application.
[mavsdk_tests] Run with -? for options
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Land on GPS lost during mission (baro height mode)
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] ../../../test/mavsdk_tests/test_multicopter_failsafe.cpp:37
[mavsdk_tests] ...............................................................................
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:186: FAILED:
[mavsdk_tests]   REQUIRE( Mission::Result::Success == result )
[mavsdk_tests] with expansion:
[mavsdk_tests]   Success == Timeout
[mavsdk_tests] terminate called after throwing an instance of 'Catch::TestFailureException'
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:186: FAILED:
[mavsdk_tests]   {Unknown expression after the reported line}
[mavsdk_tests] due to a fatal error condition:
[mavsdk_tests]   SIGABRT - Abort (abnormal termination) signal
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases: 1 | 1 failed
[mavsdk_tests] assertions: 8 | 6 passed | 2 failed
[px4         ] WARN  [mavlink] Operation timeout
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
```

julianoes, 2021-02-16:
@mrpollo thanks, I"m working through these in https://github.com/PX4/PX4-Autopilot/pull/16840

mrpollo, 2021-03-17:
Here's one more timeout on tailsitter

https://github.com/PX4/PX4-Autopilot/pull/13152/checks?check_run_id=2132856128

```sh
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] Land on GPS lost during mission (baro height mode)
[mavsdk_tests] -------------------------------------------------------------------------------
[mavsdk_tests] ../../../test/mavsdk_tests/test_multicopter_failsafe.cpp:37
[mavsdk_tests] ...............................................................................
[mavsdk_tests] ../../../test/mavsdk_tests/autopilot_tester.cpp:132: FAILED:
[mavsdk_tests]   REQUIRE( result == Action::Result::Success )
[mavsdk_tests] with expansion:
[mavsdk_tests]   Command Denied == Success
[mavsdk_tests] ===============================================================================
[mavsdk_tests] test cases: 1 | 1 failed
[mavsdk_tests] assertions: 8 | 7 passed | 1 failed
[px4         ] PX4 Exiting...
[px4         ] Exiting NOW.
[gzserver    ] 
Logfiles: 
  - logs/2021-03-17T17-33-52Z/tailsitter/Land_on_GPS_lost_during_mission_(baro_height_mode)/log-combined.log
  - logs/2021-03-17T17-33-52Z/tailsitter/Land_on_GPS_lost_during_mission_(baro_height_mode)/log-px4.log
  - logs/2021-03-17T17-33-52Z/tailsitter/Land_on_GPS_lost_during_mission_(baro_height_mode)/log-gzserver.log
  - logs/2021-03-17T17-33-52Z/tailsitter/Land_on_GPS_lost_during_mission_(baro_height_mode)/log-gzmodelspawn.log
  - logs/2021-03-17T17-33-52Z/tailsitter/Land_on_GPS_lost_during_mission_(baro_height_mode)/log-mavsdk_tests.log
    Uploading logfile '/__w/PX4-Autopilot/PX4-Autopilot/build/px4_sitl_default/tmp_mavsdk_tests/rootfs/log/2021-03-17/17_34_07.ulg' ...
    Uploaded to: https://logs.px4.io/plot_app?log=e43fe9a1-e061-4c20-b0b2-801594a512d1
--- Test case 1 of 12: 'Land on GPS lost during mission (baro height mode)' failed.
Aborting early

Results:
  - iris (not selected)
  - standard_vtol (not selected)
  - tailsitter:
     - 'Land on GPS lost during mission (baro height mode)': failed
     - 'Land on GPS lost during mission (GPS height mode)': cancelled
     - 'Continue on mag lost during mission': cancelled
     - 'Continue on baro lost during mission (baro height mode)': cancelled
     - 'Continue on baro lost during mission (GPS height mode)': cancelled
     - 'Continue on baro stuck during mission (baro height mode)': cancelled
     - 'Continue on baro stuck during mission (GPS height mode)': cancelled
     - 'Takeoff and Land': cancelled
     - 'Fly square Multicopter Missions including RTL': cancelled
     - 'Fly square Multicopter Missions with manual RTL': cancelled
     - 'Fly forward in position control': cancelled
     - 'Fly forward in altitude control': cancelled
Overall result: FAIL
Error: Process completed with exit code 1.
``

julianoes, 2021-03-18:
Thanks for collecting these @mrpollo! I will get to them at some point.

julianoes, 2021-03-22:
The ones that fail to arm should be fixed with https://github.com/PX4/PX4-Autopilot/pull/17169.

julianoes, 2021-03-25:
@mrpollo I believe some of these are now fixed. Please just re-open if you find more.

closed 2021-03-25 by julianoes