PX4/PX4-Autopilot #16760: Reuse position when local position estimate come backed

moreba1, 2021-02-05:
**Describe the bug**
when GPS fails in "no global position" and com_posctl_navl=0 drone start descends with baro information, but when local position come backed drone still don't use the position for Land and move with the wind in Land mode. 
but when we change flight mode to position it can keep a fixed position.

**To Reproduce**
Steps to reproduce the behavior:
1. Use "make px4_sitl gazebo_standard_vtol__windy" command for simulation
2. Set com_posctl_navl parameter to Land/terminate.
3. Take off to an altitude
3. Use "failure gps off" command for disabling GPS
4. After 2-second use "failure gps ok" for using GPS
5. You will see that altitude decrease in Land mode but position changes due to wind 
6. Change flight mode to POSITION or HOLD, drone keeps the position

**Expected behavior**
After GPS and local position come backed drone should use the position for Land (in com_posctl_navl=1) and changes mode to HOLD when (in com_posctl_navl=0)

bresch, 2021-02-12:
This PR (https://github.com/PX4/PX4-Autopilot/pull/16821), merged yesterday might have fixed this issue. Could you check again?
If not, could you also share a log?

moreba1, 2021-02-12:
> 
> 
> This PR (#16821), merged yesterday might have fixed this issue. Could you check again?
> If not, could you also share a log?

This problem is not resolved. 
log: https://review.px4.io/plot_app?log=c74f4326-de33-4730-809e-b8191ffa76c6

bresch, 2021-02-15:
@moreba1 In your log, the drone lands with the XY position controlled in land mode; what's not working? Is it because it takes some time to control the position after GPS is regained?
[image]

moreba1, 2021-02-15:
@bresch 
after GPS is regained drone is in descent mode still and doesn't control position, but when I change the mode to Position it controls the position.
my expected behavior is that when GPS is regained automatically flight mode changed from descent to Land.

MaEtUgR, 2021-02-15:
> my expected behavior is that when GPS is regained automatically flight mode changed from descent to Land

That totally makes sense. Let me check why that failed.