PX4/PX4-Autopilot #16763: Regression: no RC when SYS_USE_IO=0 on Pixhawk 4

bkueng, 2021-02-05:
Due to the IO module not running.
Reverting https://github.com/PX4/PX4-Autopilot/commit/d7b89ecc86a4a364f4fadf53c858fb92ac898874 fixes it.

@dagar can you have a look?

dagar, 2021-02-05:
From the context I'm piecing together that I interpreted `SYS_USE_IO` differently than you. I'll look into undoing it.

In the pixhawk 4 case what about reading RC directly FMU side? I believe it's connected to both.

LorenzMeier, 2021-02-07:
I've reverted the change in 0c6cf6cb99914925cd09f0e991031b78654e6e3e - @dagar feel free to take a stab at re-doing it.

I'm currently looking into ALWAYS taking the RC from FMU on FMUv5 / FMUv5X / FMUv6X and will test that in a separate PR. This will further dumb-down IO and make it easier to maintain stuff.

Time for old cruft and old concepts to go!

LorenzMeier, 2021-02-08:
Here is the change: https://github.com/PX4/PX4-Autopilot/pull/16780

davids5, 2021-02-08:
> I'm currently looking into ALWAYS taking the RC from FMU on FMUv5 / FMUv5X / FMUv6X and will test that in a separate PR. This will further dumb-down IO and make it easier to maintain stuff.

I think that will work on V5 but per @dagar input to the design VxX Series reused the UART6 as RC input to the FMU when the IO is not there. So it is not wired to both.