PX4/PX4-Autopilot #16767: Issues with: make px4_sitl jmavsim

YasinAl, 2021-02-05:
Hi,
When following the User guide, I keep having issues with the following command: make px4_sitl jmavsim or even DONT_RUN=1 make px4_sitl_default gazebo.
I understand that it says failed to import some stuff like toml or empy. These are installed by the script that was used to install everything. I even double checked and installed them again and the error persists.

Any ideas?

The error is as follows:

`yasin@vs ~/catkin_ws/src/PX4-Autopilot $ make clean
yasin@vs ~/catkin_ws/src/PX4-Autopilot $ make px4_sitl jmavsim
-- PX4 version: v1.11.0-rc3-1408-ga399bb9894
-- PX4 config file: /home/yasin/catkin_ws/src/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-509-gb919f3d
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yasin/catkin_ws/src/PX4-Autopilot/build/px4_sitl_default
[0/779] git submodule src/drivers/gps/devices
[7/779] Generating mixer_multirotor_6dof.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_6dof.generated.h 
cd /home/yasin/catkin_ws/src/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_6dof.generated.h
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[9/779] Generating mixer_multirotor.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor.generated.h 
cd /home/yasin/catkin_ws/src/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py -f /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor.generated.h
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[10/779] Generating mixer_multirotor_normalized.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_normalized.generated.h 
cd /home/yasin/catkin_ws/src/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --normalize -f /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometr [... cut ...] --config-files /home/yasin/catkin_ws/src/PX4-Autopilot/src/lib/battery/module.yaml /home/yasin/catkin_ws/src/PX4-Autopilot/src/drivers/gps/module.yaml /home/yasin/catkin_ws/src/PX4-Autopilot/src/modules/mavlink/module.yaml
Failed to import jinja2: No module named 'jinja2'

You may need to install it using:
    pip3 install --user jinja2

[15/779] Generating git version header
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
`

Jaeyoung-Lim, 2021-02-05:
@YasinAl As you can see in the logs, you are missing a few python packages.

Please install them as instructed in the toolchain installation: http://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#development-environment-on-ubuntu-lts-debian-linux

YasinAl, 2021-02-05:
> @YasinAl As you can see in the logs, you are missing a few python packages.
> 
> Please install them as instructed in the toolchain installation: http://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#development-environment-on-ubuntu-lts-debian-linux

@Jaeyoung-Lim They have been installed like I said in the original issue "I understand that it says failed to import some stuff like toml or empy. These are installed by the script that was used to install everything. I even double checked and installed them again and the error persists."

Any idea what the issue could be except for installing them? I have reinstalled pip and reinstalled the packages to see if it was the issue but that did not fix anything.

antonio-sc66, 2021-02-05:
@YasinAl Since you're using Ubuntu 18.04 are you sure you installed the packages in Python3? Can you test if you can import them in a Python3 interpreter?

Jaeyoung-Lim, 2021-02-06:
@YasinAl If your installations don't fix the issue, it means you have an issue with your python path and the issue is not related to this repository.

I see that you have put px4 inside your catkin workspace. Is there any chance that you have sourced the catkin workspace? If you are on ROS melodic, it will try to use python2 and break all the python paths that px4 is using.

On a side note, it is not a good practice to put px4 inside your workspace, since px4 doesn't follow the catkin directory structure

YasinAl, 2021-02-06:
> @YasinAl Since you're using Ubuntu 18.04 are you sure you installed the packages in Python3? Can you test if you can import them in a Python3 interpreter?

Actually spot on. For some reason they are not imported even though my other Python3 libraries are working. I had to reinstall them all manually (this is like the fourth time) and then they were successfully imported into a Python3 interpreter and only then did the make command work. Really weird.

closed 2021-02-06 by Jaeyoung-Lim

YasinAl, 2021-02-06:
> @YasinAl If your installations don't fix the issue, it means you have an issue with your python path and the issue is not related to this repository.
>
> I see that you have put px4 inside your catkin [...]

Maybe that was that was the issue but after reinstalling the packages and importing them somewhere else, I could run the command successfully.

Ah thanks for the side note. I will move it now!