PX4/PX4-Autopilot #16772: RPi3: LSM9DS1 Accel/Gyro and Mag Calibration Failed!

Pskoog3, 2021-02-06:
Unable to calibrate accelerometer, gyro and magnetometer for LSM9DS1 connected to Raspberry Pi 3 SPI interface. The horizon level calibration also fails.

The log is available at the following link... was unable to upload to logs.px4.io
https://drive.google.com/file/d/1UAWU1LU9kpD4hHc-K8WDeF2FWpS83VWN/view?usp=sharing
Either the sensor itself is faulty or the driver needs to be updated!

dagar, 2021-02-06:
Can you show the actual error messages? I don't see any in the log.

The z axis of the accel has a large offset, so the estimator probably isn't going to initialize until after the accel has a good calibration. The level horizon calibration depends on the attitude (vehicle_attitude publication), so it's not going to work until the estimator has initialized. 

[image]

dagar, 2021-02-06:
Your log file also shows what appears to be valid calibration data. Why do you think it failed? Can you share a screenshot of the errors?

```
CAL_ACC0_ID | 4456450
CAL_ACC0_PRIO | 50
CAL_ACC0_ROT | -1
CAL_ACC0_XOFF | 0.05530215
CAL_ACC0_XSCALE | 1.0017482
CAL_ACC0_YOFF | 0.16195849
CAL_ACC0_YSCALE | 1.0015078
CAL_ACC0_ZOFF | -1.2258157
CAL_ACC0_ZSCALE | 1.0112919
```

Pskoog3, 2021-03-05:
Daniel, here is the latest log that might show the problem...
https://drive.google.com/file/d/1zMf2fN2T35MSr7TIE3HU_hAw-lOjRNlB/view?usp=sharing

closed 2021-03-08 by Pskoog3