PX4/PX4-Autopilot #16775: Unable to get missions to move Rover forward

sardanian, 2021-02-06:
I am trying to get a rover to follow a mission plan. I have seens others are able to get this to work in SITL and recently with the changing of GND_THR_CRUISE. However I have been unsuccessful in getting the rover to move still.

I am using a Traxxas airframe and a custom mixer to allow for the rover to do reverse.

It was also mentioned in slack that changing the GND_THR_CRUISE with  sys_autostart set to 50000 so I tried that as well.
The original setting for sys_autostart was 50002 so I have presented two logs (one with sys_autostart as 50000 and the other as 50002.

I don't understand what sys_autostart does but it seems to disable my custom mixer file which is why I have given both logs:

50000: https://drive.google.com/file/d/1oxK_RkL0G986nzrYVZ0xSH7DJXrlWI39/view?usp=sharing
50002: https://drive.google.com/file/d/17TzkzmN7AoZS4dISTmoB-IDfBDEM4v3W/view?usp=sharing

Thank you for any help

LorenzMeier, 2021-02-06:
Thanks for reaching out! We're looking into this. @Jaeyoung-Lim FYI

Jaeyoung-Lim, 2021-02-06:
@sardanian Since I can see that your `GND_THR_CRUISE` is set to 0.2, Can you confirm that your vehicle can move forward with throttle value at 0.2?

And yes, that will not run your custom mixer, but will use this one: https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/mixers/rover_generic.main.mix

sardanian, 2021-02-06:
I can run it in manual mode at 20% throttle. However, in case it was an
issue I had previously tried values 10%, 20%, 30%, 40%, 50% all giving the
same results of no movement.

Thank you

Jaeyoung-Lim, 2021-02-08:
@sardanian 

I finally had time to test rover mission mode on my hardware, and mission mode works as intended on the vehicle.(using sys_autostart=50000). I was using latest master. Therefore I cannot reproduce your issue. Since https://github.com/PX4/PX4-Autopilot/pull/16297 is needed in case you have a take off waypoint at the beginning of your mission,  Can you verify which firmware version you are using? (your log points me to https://github.com/PX4/PX4-Autopilot/commit/662f177830a4084978c1f6dd4576bd568d449e0b, which should have the needed change, but not sure if flight review is showing the correct version)

sardanian, 2021-02-08:
I have have tried the latest stable and master firmwares. I am unfortunately still unable to move in mission mode. Any other thoughts or things I can test out to narrow down the issue?

Jaeyoung-Lim, 2021-02-08:
@sardanian Could you share the log of your trial with the master firmware?

What should be different with your previous log is that when entering mission mode, the actuator controls should be non-zero (assuming that you have set `GND_THR_CRUISE` larger than 0

If we can see this from your log, this means that something either related to your hardware or your mission is preventing the vehicle from moving.

sardanian, 2021-02-08:
> to your hardware or your mission is preventing the vehicle from moving. If
> not, we need to check if this is a software issue

Here is the log from today with the master firmware.

https://review.px4.io/plot_app?log=31781bbc-631e-42ff-83bf-4698b8e340f8

Jaeyoung-Lim, 2021-02-08:
@sardanian Your log shows that you were in mission mode for less than 1 second. Is this expected?

sardanian, 2021-02-09:
Yes. I tried it and nothing happened. I can send another log, but same
results will happen. This was the last log I did, but I have a few others
with longer mission times if needed. All the same result of no movement
though.

Thank you

Jaeyoung-Lim, 2021-02-09:
@sardanian In order to verify your mission, have you tried running your mission in SITL?

sardanian, 2021-02-10:
I’ll try it out. I’m assuming I need to use my current GPS coordinates and
go THR same path. I haven’t changed the default SITL GPS but I assume there
is. If it’s easy, can you point me in a direction to alter that? If not,
I’ll search around.

closed 2021-02-11 by LorenzMeier