PX4/PX4-Autopilot #16777: the problem of "make px4fmu-v2_default"

Reality2Ideal, 2021-02-07:
When i begen to compile Firmware,the problem occured.

PX4 console:
`LX@LAPTOP-FH0SSNDV /g/px4/Firmware
$ make px4fmu-v2_default
Makefile:496: *** "Make target px4fmu-v2_default not found. It either does not exist or px4fmu-v2_default cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.`

Eclipse:
[image]

This makes me very anxious!

I want to know why and how to solve this problem.

TSC21, 2021-02-07:
It's `make px4_fmu-v2_default`

closed 2021-02-07 by TSC21

Reality2Ideal, 2021-02-07:
> make px4_fmu-v2_default

really?
PX4 console:
`$ make px4_fmu-v2_default
/bin/sh: cmake: command not found
rm: cannot remove directory `/g/px4/Firmware/build/px4_fmu-v2_default': Permission denied
make: *** [px4_fmu-v2_default] Error 1`
Eclipse:
`18:52:42 **** Build of configuration Default for project Firmware ****
make px4_fmu-v2_default 
/bin/sh: cmake: command not found
rm: cannot remove directory `/g/px4/Firmware/build/px4_fmu-v2_default': Permission denied
make: *** [px4_fmu-v2_default] Error 1`

TSC21, 2021-02-07:
> really?
> PX4 console:
> `$ make px4_fmu-v2_default /bin/sh: cmake: command not found rm: cannot remove directory `/g/px4/Firmware/build/px4_fmu-v2_default': Permission denied
> make: *** [...]

Yes. The above shows you clearly have some problems on your toolchain. I advise you too check the Getting Started guide: https://docs.px4.io/master/en/dev_setup/getting_started.html

Reality2Ideal, 2021-02-07:
> Yes. The above shows you clearly have some problems on your toolchain. I advise you too check the Getting Started guide: https://docs.px4.io/master/en/dev_setup/getting_started.html

Thank you.I will have a look!

LorenzMeier, 2021-02-07:
It looks like you might have run some commands with sudo, resulting in permission issues on your file system. You will want to delete these directories and make sure to not run sudo.

Reality2Ideal, 2021-02-07:
> It looks like you might have run some commands with sudo, resulting in permission issues on your file system. You will want to delete these directories and make sure to not run sudo.

I'm sure I didn't run commands with sudo,because I install it on Windows .

LorenzMeier, 2021-02-07:
Ok - then it's a windows toolchain issue.