PX4/PX4-Autopilot #16778: Run the offboard mode under the optical track

AllenHuangDongyue, 2021-02-07:
Hey guys, 

I am using the optical track for the external location, and I want my drone to hover at a point by using the offboard mode. And I am using radio telemetry to send the mavros message.
However, when the drone arrived at the targeted position, it will keep rising.
Is there anyone who knows the reason?

Btw, I uploaded the log and changed its name into .txt.
Thank you!
Allen
[log_129_2021-2-7-21-40-02.ulg.txt](https://github.com/PX4/PX4-Autopilot/files/5939212/log_129_2021-2-7-21-40-02.ulg.txt)

closed 2021-04-28 by AllenHuangDongyue

ashwinlele, 2021-05-19:
Hi Jaeyoung-Lim, I am using the direct UART connection between FMUK66 and Jetson nano for MAVLink messages. I think it is sufficiently fast. Still, the drone starts rising on switching to offboard mode. Can you please tell me a possible solution to this?

Thanks!

Jaeyoung-Lim, 2021-05-19:
@ashcharles Then I believe your issue is not the same as the issue posted here. Please post a new issue, with sufficient information (your setup, how you are running things, how it is failing)

ashcharles, 2021-05-19:
> @ashcharles Then I believe your issue is not the same as the issue posted here. Please post a new issue, with sufficient information (your setup, how you are running things, how it is failing)

I think this comment is intended for @ashwinlele not me :)