PX4/PX4-Autopilot #16787: fixedwing runwaytakeoff (RWTO) broken

dagar, 2021-02-08:
Last working in CI here. https://github.com/PX4/PX4-Autopilot/pull/16564

@ryanjAA any details you could add?

ryanjAA, 2021-02-08:
Current master seems to be taking off somewhere between trim and max speed. Also, the standard sitl plane seems to not listen to the take off pitch parameter (when set to a hard number or 0) and the qgc pitch number in the mission. It always goes above 20 deg and the moment it does it flips up to 90 deg. The only way to continue to the mission is to click on the takeoff waypoint or any other waypoint and tell it to go to it and the mission then is fine. 

I’ll do it again a few times and look at the log for exactly at what speed it’s pitching up too see if the pitch up takeoff scale parameter is playing into this.

sfuhrer, 2021-02-12:
Current master seems completely broken. With https://github.com/PX4/PX4-Autopilot/pull/16792 though it's able to takeoff. Some findings still though:

It's wrong in my eyes that it doesn't consider FW_THR_MAX here, but always ramps up to 100% throttle --> should fix that.
[image]

For some reason it seems to be glued to the ground during the Climbout state, even though it gives a lot of throttle and full elevator deflection up. I found that by slightly giving some flaps during the takeoff, this phase can be shortened (here I apply 20% flaps during takeoff). Is that a bad modelling of some ground contact force? @Jaeyoung-Lim would you have any clue?
[image]
The filled integrator then lets the pitch overshoot.

gvo4davsys, 2021-02-26:
Hello, I think I am also experiencing something related to this issue and figured I'd share what I've seen. I am running the current master of PX4, and when running Gazebo SITL simulations with the standard fixed wing, takeoff behavior has been broken when running with a ROS node. If I run `make px4_sitl_default gazebo_plane`, thus launching without ROS node, then using missions, or taking off via commander, works perfectly fine. If I run `DONT_RUN=1 make px4_sitl_default gazebo_plane` and then launch Gazebo via ROS launch commands then attempting to takeoff via missions or commander results in the aircraft flipping through the air and crashing immediately. I have also tried pegging the throttle and pitch actuator command via mavros actuator_control message. When doing this I experience the same "glued to the ground" behavior seen by @sfuhrer . Visual inspection of the plane model in the Gazebo GUI shows that the elevators do in fact move as expected, but the aircraft does not takeoff. This behavior is the exact same for the standard plane_catapult as well. Hope this information is helpful/relevant

dagar, 2021-02-26:
@ryanjAA Where do we stand with current master after https://github.com/PX4/PX4-Autopilot/pull/16792 has merged?

ryanjAA, 2021-02-28:
@dagar fixed wing takeoff does work, sort of. It just doesn't listen to the minimum takeoff scale for pitch vs min airspeed when below 1 as far as I can tell. If you set it to 2 (double min speed), it listens. 1.3x and it listens but below 1 it doesn't seem to.

It also does act differently depending on the minimum takeoff pitch as well but it doesn't spin out of control like before and it does takeoff.

Takeoff scale at:

[1.3 and 10 deg min pitch](https://review.px4.io/plot_app?log=83419a73-5170-4e0d-b14b-09c85b8f5957)
[0.85 and 10 deg min pitch](https://review.px4.io/plot_app?log=e2c4de03-88c6-4690-b92d-60b5b428a781)
[0.85 and 20 deg min pitch](https://review.px4.io/plot_app?log=e9f748d5-d3a0-4d20-814e-8c1064b8ea2a)
[2.0 and 20 deg min pitch](https://review.px4.io/plot_app?log=315ac2dd-91cc-4613-8eec-846be4a39271)

gvo4davsys, 2021-03-01:
@ryanjAA I changed the RWTO_AIRSPEED_SCL to 200% and the FW_TKO_PITCH_MIN to 20, and it still spins out of control when taking off while launched with ROS. I had a branch on a forked repository of Tools/sitl_gazebo that was at commit [15306c1cb1ba3d0723ab29c3e59444abec53e85b](https://github.com/PX4/PX4-SITL_gazebo/tree/15306c1cb1ba3d0723ab29c3e59444abec53e85b) and that works fine with the rest of PX4 master. So maybe something in Tools/sitl_gazebo isn't playing nice with ROS?