PX4/PX4-Autopilot #16790: Thoneflow does not provide good estimates

int-smart, 2021-02-08:
Hi
I have been using Thoneflow to interface the Thoneflow 3901UY optical camera with PX4. I have a Benewake TFMini Plus distance sensor attached to the setup. I am able to get the raw readings in the MavLink Inspector. However I am observing that the flow velocities from the camera are very small as compared to the vehicle_angular_velocity. 

Steps to reproduce the behavior:
1. Drone switched on with TFMini Plus and Thoneflow 3901UY.
2. Took off.

I expect to see the angular flow around x from optical flow to be similar to the vehicle_angular_velocity. I am comparing these considering #12148.

Log File: https://logs.px4.io/plot_app?log=fd18a2a5-e78f-4421-a0d7-b0c86532e08b
[image]

Drone is a quadcopter with TFMini Plus facing down and optical flow camera mounted so that the small notch on ThoneFlow 3901UY faces towards negative x direction of the drone.

Thanks

bresch, 2021-02-09:
For better optical flow debugging, you can log `estimato_optical_flow_vel` at full rate (change https://github.com/PX4/PX4-Autopilot/blob/9f00fd279629b5782f2831e17cdb6826e9dc0dcf/src/modules/logger/logged_topics.cpp#L143; set 200->0). This contains the pre-processed optical flow data and can be compared with estimated velocity and gyro data.

jinger26, 2021-02-10:
@bys1123  can you comment?

closed 2021-02-25 by dagar

bys1123, 2021-02-25:
@int-smart sorry, the current driver is for old hardware version, now I updated this to latest, you could try master, it should works well.

PR #16946 (2021-02-25): Update the data timespan in the driver of ThoneFlow
two year ago, ThoneFlow-3901UY updated the framerate from 66 (3901U) to 95fps. so the timespan changed.
old version are not much in the market, so chage this value to new version.
Fix #16790