PX4/PX4-Autopilot #16801: Estimator divergence after landing with mag stuck

julianoes, 2021-02-09:
The SITL integration test which simulates the mag being stuck sometimes fails because landing is not detected.

We see that the estimated vertical velocity `vehicle_local_position.vz` has an offset after landing which then causes the land detector to stay "in-air".

@priseborough @bresch I don't understand all of the mechanisms at play here that lead to this result and I would appreciate any pointers.

Logs:
https://review.px4.io/plot_app?log=3c5f784f-7bd8-4a51-b5ff-cf7edf1ac709

On the plots below you can see how attitude as well as velocity estimate diverge once on ground.
[image]

bresch, 2021-02-16:
During a debugging session with @priseborough , we found that the issue is the following:

The mag is correctly rejected in air, however, as soon as the drone touches down, the landed flag is set and that switches the EKF in "landed mode" (`in_air` == `false`). When not `in_air`, the EKF fuses the mag heading with a limited innovation even if the test is failing, because that could be due to an initial large gyro bias or a change in ground magnetic field.
The logic is here: https://github.com/PX4/PX4-ECL/blob/d4258cc66d24c6871688ae83f42de3b46c96d033/EKF/mag_fusion.cpp#L646-L656

The yaw is then pulled away and the velocity estimate is wandering around; the land detector switches then out of `landed`.
A potential fix is then to delay that forced fusion logic on ground to be only active after the land detector had the change to disarm the drone.

[image]

LorenzMeier, 2021-02-16:
What is the follow up action here?

bresch, 2021-02-16:
> What is the follow up action here?

I'll add a delay

closed 2021-02-18 by bresch